fkie-cad/FACT_core #892: Internal Server Error when trying to access upload page

yhtiee, 2022-10-27:
i get an internal server error when trying to access the upload page from the home page

jstucke, 2022-11-07:
Hi and sorry for the late response. What do you see in the command line log? Are there any error messages / stack traces?

code2319, 2022-11-11:
I have the same error, log:
```
[2022-11-11 20:16:21,890] ERROR in app: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.10/dist-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/components/io_routes.py", line 43, in get_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "/home/user/Downloads/FACT_core-master/src/./intercom/front_end_binding.py", line 34, in get_available_analysis_plugins
    raise Exception('No available plug-ins found. FACT backend might be down!')
Exception: No available plug-ins found. FACT backend might be down!
[2022-11-11 20:16:21][app][ERROR]: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.10/dist-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/components/io_routes.py", line 43, in get_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "/home/user/Downloads/FACT_core-master/src/./intercom/front_end_binding.py", line 34, in get_available_analysis_plugins
    raise Exception('No available plug-ins found. FACT backend might be down!')
Exception: No available plug-ins found. FACT backend might be down!
[pid: 22581|app: 0|req: 12/63] 127.0.0.1 () {42 vars in 717 bytes} [Fri Nov 11 20:16:21 2022] GET /upload => generated 265 bytes in 63 msecs (HTTP/1.1 500) 2 headers in 99 bytes (1 switches on core 1)
```

jstucke, 2022-11-14:
Was there some other error earlier in the logs? It seems the web frontend is missing data here that should be passed from the backend but somehow that never happened, hence the error. If not, is the backend displayed as "online" on the `/system_health` page and can you analyze firmware using the "Run additional analysis" feature from the analysis page?

code2319, 2022-11-14:
Was there some other error earlier in the logs? - no internal routing support, rebuild with pre support (after detected binary path).
is the backend displayed as "online" on the `/system_health` page - Internal server error
```bash
[2022-11-11 20:16:21][app][ERROR]: Exception on /system_health [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.10/dist-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.10/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/security/decorator.py", line 11, in decorated_view
    return fn(*args, **kwargs)
  File "/home/user/Downloads/FACT_core-master/src/./web_interface/components/io_routes.py", line 43, in get_upload
    analysis_plugins = sc.get_available_analysis_plugins()
  File "/home/user/Downloads/FACT_core-master/src/./intercom/front_end_binding.py", line 34, in get_available_analysis_plugins
    raise Exception('No available plug-ins found. FACT backend might be down!')
Exception: No available plug-ins found. FACT backend might be down!
```
"Run additional analysis" feature from the analysis page? - where is it?

jstucke, 2022-11-15:
> "Run additional analysis" feature from the analysis page - where is it?

It's the page with the analysis and meta data that you see when you click on a firmware / file from the start page or from the search results. On the left where you can select the analysis at the bottom there is a button with "Run additional analysis" written on it. If you click in it, you can run analysis plugins just for this single file.

> Exception: No available plug-ins found. FACT backend might be down!

Either the backend seems to have problems when starting or the communication between the frontend and the backend over Redis does not work.

Can you post the full log of the backend that you see when you start it separately with `src/start_fact_backend.py`? There may still be an error hidden somewhere.

Does `service redis status` say "active (running)"?

jstucke, 2022-11-16:
Those are a lot of errors. It seems some dependencies are missing. How did you install FACT? Did you follow the [INSTALL.md](https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.md#simple-one-system-setup)? I would suggest rerunning `python3 install.py` from the `src` directory.

chenjianquan7, 2023-01-05:
it because code has a lot of error