fkie-cad/FACT_core #907: Accessing dashboard via IP from remote host within same network

senthilkumarmoorthy, 2022-11-18:
Hello, as per the INSTALL.md before running src/install.py I made exactly like the instruction said but additionally I edited FACT_core/src/config/uwsgi_config.ini and appended my local host IP as show below and then I ran src/install.py –nginx.
[image]
Post installation I started ./start_all_installed_fact_components and in the backend I also verified by running netstat -tlpen to confirm redis-server and fact_core running status and found this
[image]
So, then upon accessing the dashboard using http://192.168.1.20:5000/ I get Can’t reach the page from a host 192.168.1.21 (Which is a Windows 11 machine running on the same network). 
While I try to access the dashboard from being inside the 192.168.1.20 host it works fine. Need someone help to get the URL accessed from any machine within the same network.

jstucke, 2022-11-21:
Hi, you don't really need NGINX if your goal is to make FACT available in your local network. You can set "0.0.0.0" as IP in the uWSGI config, then your FACT instance should be reachable from other systems in the same LAN.

> http-socket = 0.0.0.0:5000

senthilkumarmoorthy, 2022-11-22:
> Hi, you don't really need NGINX if your goal is to make FACT available in your local network. You can set "0.0.0.0" as IP in the uWSGI config, then your FACT instance should be reachable from other systems in the same LAN.
> 
> 
> 
> > http-socket = 0.0.0.0:5000

I will try this.

senthilkumarmoorthy, 2022-11-23:
> Hi, you don't really need NGINX if your goal is to make FACT available in your local network. You can set "0.0.0.0" as IP in the uWSGI config, then your FACT instance should be reachable from other systems in the same LAN.
> 
> > http-socket = 0.0.0.0:5000

No luck.
[image]

jstucke, 2022-11-23:
The netstat output looks correct. "0.0.0.0:5000" means that your FACT instance should be reachable from other devices in the network using the IP of the system and port 5000 (i.e. http://192.168.1.20:5000/  if that hasn't changed). Could there be anything else that might interfere (a firewall maybe)?

Do you see any errors in the log that is printed when starting FACT? There should be something like 

> uwsgi socket 0 bound to TCP address 0.0.0.0:5000

and you should be able to reach it from the device itself (e.g. `curl http://localhost:5000/ `).

jstucke, 2022-11-28:
That's great to hear. I will close this issue then. Feel free to reopen it if you have further problems/questions.

closed 2022-11-28 by jstucke