fkie-cad/FACT_core #913: Problem when running docker build

qy201706, 2022-11-26:
## error logs

[
[image]
](url)

**How can I fix it?please help me.**

qy201706, 2022-11-26:
my Ubuntu is 20.04 on WSL2, docker has been installed.

jstucke, 2022-11-28:
Hi, FACT support in WSL is still somewhat experimental. The cause of the error seems to be a connection issue. Are you working behind a proxy? What happens if you try to run the failing command directly in the terminal outside of docker. Does it work?

```bash
curl https://raw.githubusercontent.com/danielmiessler/SecLists/master/Passwords/Common-Credentials/10k-most-common.txt
```

It could also be a temporary thing. What happens if you try to run the installation again? You can also try to run the installation of the problematic plugin directly by running this from the `src` directory:

```bash
python3 plugins/analysis/users_and_passwords/install.py
```

jstucke, 2024-11-13:
Since there has been no update to this issue for quite a while I will close it. Feel free to reopen it if you have additional questions.

closed 2024-11-13 by jstucke