fkie-cad/FACT_core #931: Failed to compile yara test signatures

redghostplus, 2022-12-14:
I installed FACT via venv but got the following error. Looks like there was an error during creation. I checked my signature folder and it was empty. I don't know if I'm missing something?

[2022-12-14 10:25:20][backend][INFO]: Installing linter plugin.
[2022-12-14 10:26:04][backend][INFO]: Finished installing linter plugin.

[2022-12-14 10:26:04][backend][INFO]: Creating firmware directory
Create signature directory /home/redghost/RedPlus/tools/FACT/src/analysis/signatures
Compile signatures in /home/redghost/RedPlus/tools/FACT/src/plugins/analysis/software_components/signatures
[ERROR] Creation of /home/redghost/RedPlus/tools/FACT/src/analysis/signatures failed !!
Compile signatures in /home/redghost/RedPlus/tools/FACT/src/plugins/analysis/known_vulnerabilities/signatures
[ERROR] Creation of /home/redghost/RedPlus/tools/FACT/src/analysis/signatures failed !!
Compile signatures in /home/redghost/RedPlus/tools/FACT/src/plugins/analysis/crypto_hints/signatures
[ERROR] Creation of /home/redghost/RedPlus/tools/FACT/src/analysis/signatures failed !!
Compile signatures in /home/redghost/RedPlus/tools/FACT/src/plugins/analysis/crypto_material/signatures
[ERROR] Creation of /home/redghost/RedPlus/tools/FACT/src/analysis/signatures failed !!
Traceback (most recent call last):
  File "/home/redghost/RedPlus/tools/FACT/src/install.py", line 214, in <module>
    install()
  File "/home/redghost/RedPlus/tools/FACT/src/install.py", line 182, in install
    install_fact_components(args, distribution, none_chosen, skip_docker)
  File "/home/redghost/RedPlus/tools/FACT/src/install.py", line 202, in install_fact_components
    backend(skip_docker, distribution)
  File "/home/redghost/RedPlus/tools/FACT/src/install/backend.py", line 62, in main
    raise InstallationError('Failed to compile yara test signatures')
helperFunctions.install.InstallationError: Failed to compile yara test signatures

closed 2022-12-15 by redghostplus

blinkybob, 2023-09-20:
@redghostplus  what was the problem? I am running into the same...thanks

blinkybob, 2024-01-25:
solution for me was
```
sudo echo "/usr/local/lib" >> /etc/ld.so.conf
sudo ldconfig
```
the loader couldn't find the libs...