fkie-cad/FACT_core #938: Don't kill the whole processgrop on CTRL-C

maringuu, 2022-12-22:
Just sending SIGKILL to the whole processgroup after shutting down was (kind of) successful is confusing.
As the terminal will always show `killed` and the exit code will not be zero.

See also https://github.com/fkie-cad/FACT_core/pull/809#issuecomment-1268554211

jstucke, 2023-01-02:
We had issues with subprocesses continuing to run as "zombie processes" after shutdown when exceptions occurred at certain places in the code. This can lead to all kinds of interesting problems if a "shadow" frontend or backend is still running in the background. Killing the process group was the only halfway reliable way to fix this but maybe you can think of a better way.

jstucke, 2023-04-18:
Should be resolved in https://github.com/fkie-cad/FACT_core/pull/953

closed 2023-04-18 by jstucke