fkie-cad/FACT_core #958: Postgres url fails when password contains `@`

lrcarp17, 2023-01-13:
I was running the DB migration, and it kept failing. After tracking down the problem, it led me to my password, `testest@r1`(not my real password but a comparative example). The error output i got was `swiper@factor:/FACT_core$ sudo python3 src/migrate_db_to_postgresql.py
ERROR:root:Database error when trying to read from the database: (psycopg2.OperationalError) could not translate host name "r1@localhost" to address: Name or service not known`. I believe the issue is in the `DbConnection` class in `src/storage/db_connection.py`. The line responsible for this is `engine_url = f'postgresql://{user}:{password}@{address}:{port}/{database}'`. My workaround was putting `password = password.replace("@","%40")`. I know that's not a great fix but i just needed it to work tbh.

jstucke, 2023-01-16:
We didn't anticipate this problem but yeah it's kind of obvious what the problem is: username and password need to be URL encoded. #959 should probably fix this

jstucke, 2023-01-17:
#959 should have fixed this. Can you confirm this? If it doesn't work, please reopen the issue.

PR #959 (2023-01-17): storage/db_connection: Fix engine url
We did not properly escape special characters.
This is exactly what `sqlalchemy.URL` is for.

Closes #958
closed 2023-01-17 by jstucke