fkie-cad/FACT_core #964: crypto_hints plugin data is not  JSON serializable which causes migration failur

lrcarp17, 2023-01-19:
While migrating, I came across the following error. 
```
firmwares ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━  88% 0:00:03
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 1800, in _execute_context
    context = constructor(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/default.py", line 1121, in _init_compiled
    param = {
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/default.py", line 1122, in <dictcomp>
    key: processors[key](compiled_params[key])
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/sql/sqltypes.py", line 2664, in process
    serialized = json_serializer(value)
  File "/usr/lib/python3.8/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.8/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type bytes is not JSON serializable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/FACT_core/src/storage/db_interface_base.py", line 51, in get_read_write_session
    session.commit()
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 1451, in commit
    self._transaction.commit(_to_root=self.future)
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 829, in commit
    self._prepare_impl()
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 808, in _prepare_impl
    self.session.flush()
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 3444, in flush
    self._flush(objects)
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 3584, in _flush
    transaction.rollback(_capture_exception=True)
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/util/langhelpers.py", line 70, in __exit__
    compat.raise_(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/util/compat.py", line 208, in raise_
    raise exception
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/session.py", line 3544, in _flush
    flush_context.execute()
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/unitofwork.py", line 456, in execute
    rec.execute(self)
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/unitofwork.py", line 630, in execute
    util.preloaded.orm_persistence.save_obj(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/persistence.py", line 245, in save_obj
    _emit_insert_statements(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/orm/persistence.py", line 1097, in _emit_insert_statements
    c = connection._execute_20(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 1705, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/sql/elements.py", line 333, in _execute_on_connection
    return connection._execute_clauseelement(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 1572, in _execute_clauseelement
    ret = self._execute_context(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 1806, in _execute_context
    self._handle_dbapi_exception(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 2124, in _handle_dbapi_exception
    util.raise_(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/util/compat.py", line 208, in raise_
    raise exception
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py", line 1800, in _execute_context
    context = constructor(
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/default.py", line 1121, in _init_compiled
    param = {
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/default.py", line 1122, in <dictcomp>
    key: processors[key](compiled_params[key])
  File "/usr/local/lib/python3.8/dist-packages/sqlalchemy/sql/sqltypes.py", line 2664, in process
    serialized = json_serializer(value)
  File "/usr/lib/python3.8/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.8/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
sqlalchemy.exc.StatementError: (builtins.TypeError) Object of type bytes is not JSON serializable
[SQL: INSERT INTO analysis (uid, plugin, plugin_version, system_version, analysis_date, summary, tags, result) VALUES (%(uid)s, %(plugin)s, %(plugin_version)s, %(system_version)s, %(analysis_date)s, %(summary)s::VARCHAR[], %(tags)s, %(result)s)]
[parameters: [{'analysis_date': 1668629024.0888274, 'result': {'plugin_used': 'NOP', 'info': 'unpacking skipped', 'number_of_excluded_files': 0, 'number_of_unpacke ... (504231 characters truncated) ... 2a0d767b31eda754d5ee938ebcf99446f870ff890611f6627_111404', 'summary': [], 'plugin': 'cwe_checker', 'plugin_version': '0.5.1', 'system_version': None}]]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "migrate_db_to_postgresql.py", line 396, in <module>
    main()
  File "migrate_db_to_postgresql.py", line 294, in main
    migrated_fw_count = migrator.migrate_fw(query={}, root=True, label='firmwares')
  File "migrate_db_to_postgresql.py", line 344, in migrate_fw
    self._migrate_single_object(firmware_object, parent_uid, root_uid)
  File "migrate_db_to_postgresql.py", line 364, in _migrate_single_object
    self.postgres.insert_object(firmware_object)
  File "/FACT_core/src/storage/db_interface_backend.py", line 34, in insert_object
    self.insert_firmware(fw_object)
  File "/FACT_core/src/storage/db_interface_backend.py", line 75, in insert_firmware
    print [[fo_entry, firmware_entry, *analyses]]
  File "/usr/lib/python3.8/contextlib.py", line 120, in __exit__
    next(self.gen)
  File "/FACT_core/src/storage/db_interface_base.py", line 55, in get_read_write_session
    raise DbSerializationError() from err
storage.db_interface_base.DbSerializationError
```
After a lot of digging I traced down the analysis causing the problem, which was `crypto_hints`. By adding 
```
import json
for plugin_name, analysis_data in firmware.processed_analysis.items():
        print(plugin_name)
        print(analysis_data)
        json.dumps(analysis_data)
```
in /FACT_core/src/storage/db_interface_backend.py after line 67.  This revealed that the `crypto_hints` data was not serializable. The printout of that dictionary can be seen at the bottom of this issue. Not sure what a proper fix for this is but for now, I'm just going to sanitize this dict with some hard code.

[cryto_hints_dump.txt](https://github.com/fkie-cad/FACT_core/files/10459345/cryto_hints_dump.txt)

lrcarp17, 2023-01-19:
I feel like the best solution is to ensure data is sanitized in `create_analysis_entries`  from [src/storage/entry_conversion.py](https://github.com/fkie-cad/FACT_core/blob/master/src/storage/entry_conversion.py)

lrcarp17, 2023-01-19:
sanitzing bytes in `_sanitize_value` from [src/storage/entry_conversion.py](https://github.com/fkie-cad/FACT_core/blob/master/src/storage/entry_conversion.py) is probably the solution

lrcarp17, 2023-01-19:
so after futher review, i noticed the sanitize functions in my case `_sanitize_list` is missing a handler for tuples.

lrcarp17, 2023-01-19:
Okay i fixed her is my patch

```
diff --git a/src/storage/entry_conversion.py b/src/storage/entry_conversion.py
index b4313fe6..6bc8c277 100644
--- a/src/storage/entry_conversion.py
+++ b/src/storage/entry_conversion.py
@@ -115,12 +115,18 @@ def _sanitize_value(analysis_data: dict, key: str, value):
         analysis_data[key] = value.replace('\0', '')
     elif isinstance(value, list):
         _sanitize_list(value)
+    elif isinstance(value, bytes):
+        analysis_data[key] = value.decode()
 
 
 def _sanitize_key(analysis_data: dict, key: str):
     if '\0' in key:
         analysis_data[key.replace('\0', '')] = analysis_data.pop(key)
 
+def _sanitize_tuple_and_return(elm : tuple):
+    tpl_lst = list(elm)
+    san_list = _sanitize_list(tpl_lst)
+    return tuple(san_list) 
 
 def _sanitize_list(value: list) -> list:
     for index, element in enumerate(value):
@@ -128,11 +134,17 @@ def _sanitize_list(value: list) -> list:
             sanitize(element)
         elif isinstance(element, str) and '\0' in element:
             value[index] = element.replace('\0', '')
+        elif isinstance(element, bytes):
+            value[index] = element.decode()
+        elif isinstance(element, tuple):
+            value[index] = _sanitize_tuple_and_return(element)      
     return value
 
 
 def create_analysis_entries(file_object: FileObject, fo_backref: FileObjectEntry) -> list[AnalysisEntry]:
-    return [
+    result = []
+    for plugin_name, analysis_data in file_object.processed_analysis.items():
+        result.append(
         AnalysisEntry(
             uid=file_object.uid,
             plugin=plugin_name,
@@ -143,9 +155,9 @@ def create_analysis_entries(file_object: FileObject, fo_backref: FileObjectEntry
             tags=analysis_data.get('tags'),
             result=get_analysis_without_meta(analysis_data),
             file_object=fo_backref,
-        )
-        for plugin_name, analysis_data in file_object.processed_analysis.items()
-    ]
+        ))
+        
+    return result
 
 
 def analysis_entry_to_dict(entry: AnalysisEntry) -> dict:

```

jstucke, 2023-01-20:
Hi and thank you for your debugging work on this. I tried to add all fixes (including the patch) in #966 
Converting the tuples in the analysis results to lists should be enough, because they are converted to lists during JSON conversion anyway. Would you mind trying it out to see if it runs through in your case?

PR #966 (2023-02-07): Migration bugfixes
- fixed a bug with tuples in analysis results not being sanatized
- added exception handling for custom classes occurring in pickled ("sanitized") results
- split migration of `FileObject`s and analyses
  - the migration should continue if single analyses cannot be migrated

resolves #964
closed 2023-02-07 by jstucke