fkie-cad/FACT_core #972: uwsgi use cpu 100%，how to solve?

chenjianquan7, 2023-02-01:
[image]
web  responses very slow，how to solve?

dorpvom, 2023-02-16:
Hi chenjianquan7,

could you elaborate on your observation? FACT is mostly a server-side application, doing most of the computational work in the backend. Thus, uwsgi in hosting the frontend code should not do much aside from request processing and response generation from backend-supplied data.

- Do you perform a specific task on the observed system?
- Is it a single user FACT or do you host for a team (and multiple parties might query FACT at the same time)?
- Are you possibly running REST API scripts against that FACT?

All such information might make narrowing down the issue more feasible.

jstucke, 2024-11-13:
Since there has been no response for over a year I will close this issue. Please reopen it if the problem still persists.

closed 2024-11-13 by jstucke