fkie-cad/FACT_core #989: About yara matching software component version

gaohang, 2023-03-02:
### Your question

why FACT uses ghidra extracting version format string instead of yara matching directly?
yara can match by regex rule. I can not understand key_string. Could you give some guidence? thank you.

gaohang, 2023-03-02:
why version strings are in xref basic block?

> FACT uses mostly YARA to directly match the version. Some programs store the version in a separate string and use string formatting for their version output (e.g. "xxx version %s" or "yyy version %d.%d.%d"). Matching the version directly is usually not robust enough because of too many false positives. We use Ghidra in these cases to try to find the string (or number) that is formatted into the string.
> 
> The `key_string`s are the format strings from the signature (only the signatures where `format_string = true` is set in the meta data). The idea is to find the basic block where these strings are used and look for the version strings from there.

Enkelmann, 2023-03-03:
To be more precise we look for the call that prints the version string (and not only the basic block where it happens). And then we follow the other parameters to that call (which correspond to the version numbers in e.g. "yyy version %d.%d.%d") back to their sources using the xrefs of Ghidra.