fkie-cad/FACT_core #991: Reconsider `start_fact_db.py`

maringuu, 2023-03-06:
It does not do anything than check if postgres is running and provide statistics.

dorpvom, 2023-03-14:
I recently brought up that we changed our system model when switching database engines. We previously ran on a non-standard port to not encumber previously existing environments. Going back to this model would re-introduce starting the database engine (on a non-standard port) in this script.
We should at least discuss this towards a final decision before removing the script.

dorpvom, 2023-03-14:
Also, dropping system statistics for the backend opens the question if these are relevant. If the answer is no, it is arguable, that the system stats for the other parts are irrelevant as well. I'd always argue for consistency in these matters. Ideas to make this consistent without this script can be discussed.

maringuu, 2024-02-19:
> If the answer is no, it is arguable, that the system stats for the other parts are irrelevant as well.

I'd argue that they are irrelevant in the sense that it is not FACT's responsibility to provide statistics about the server usage.
While I have not used it myself, it think using [Prometheus](https://prometheus.io/) is the right choice.