fkie-cad/FACT_core #1004: How to troubleshoot firmware submission failure with high failure rate due to possible packet loss on the platform?

chenjianquan7, 2023-03-29:
### Your question

I pushed a firmware to the platform, but it was not found during the search process. It seems that the submission was not successful, possibly due to packet loss, as the failure rate is quite high. How can I resolve this issue?

jstucke, 2023-03-29:
What version are you using (release or latest master branch)? 
Are there any errors in the terminal log?
What exactly do you mean with "search process"?

chenjianquan7, 2023-03-31:
> 您使用的是什么版本（发行版或最新的主分支）？ 终端日志中是否有任何错误？ “搜索过程”到底是什么意思？

v4.0.2，master
no error，you can push a firm ,then click firm address ,it is null mostly

jstucke, 2023-03-31:
I'm not sure if I understand your problem correctly. If you upload a firmware, do you see files being analyzed in the backend log on the command line? Does the firmware show up under "currently analyzed firmware" on the system page ("Info" -> "System")?

Maybe you could try switching to the current master version. You will probably need to rerun the installation then, though.