fkie-cad/FACT_core #1013: Does Fact run any ubuntu server not a window

ramazankoyuturk, 2023-04-10:
Hi, 

İ want to run Fact in the powerfull EC2. This server is ubuntu server and there is not any windows. So i must use command line interface. 
How can i analysis any firmware using ubuntu terminal?

Thanks

dorpvom, 2023-04-12:
Also: Make sure to configure FACT (using the `src/config/main.cfg` or by creating an own config file) to make proper use of resources. I could give you an example configuration for a server with 64 cores and 256 GB RAM.

We'll probably write a blog post on scaling FACT in the near future, since our recent work allows for more configuration and performance.

jstucke, 2023-04-18:
> How can i analysis any firmware using ubuntu terminal?

> since FACT has a [REST API](https://github.com/fkie-cad/FACT_core/wiki/Rest-API) you should be able to script all interaction with FACT an run it from a command line interface. Querying for results gives you the information in JSON.

I just want to add (in case this might not be clear) that you can of course access the web GUI of FACT from every other system in the network if you adjust the IP configuration of the web server and upload firmware from there.