fkie-cad/FACT_core #1015: it does not work when run install.py

chenjianquan7, 2023-04-23:
### Your question

[image]
it is not run ,stop at it

jstucke, 2023-04-24:
Are you sure it does not work? You can ignore the warning message. The log level of the installer was erroneously set to "WARNING" as default but you can run the installation with `./src/install.py -L DEBUG` to get more feedback during the installation. It can take while for it to complete (>15 minutes).

BREAKJADE, 2023-06-19:
[2023-06-19 08:21:41][install][ERROR]: Pip package sqlalchemy==1.4.43 could not be installed:
ERROR: Could not find a version that satisfies the requirement sqlalchemy==1.4.43 (from versions: none)
ERROR: No matching distribution found for sqlalchemy==1.4.43