fkie-cad/FACT_core #1041: process cannot run finish?

chenjianquan7, 2023-05-17:
### The FACT version you are using

4.1

### Your question

it analysis beyond 24h
[image]
some errors 
[image]
[image]
[image]
 the reason is unpack or analysis ?
why happend  some times?

jstucke, 2023-06-12:
Sorry for the late response. You can ignore the "generic carver fallback" warnings (they only mean that unpacking a specific file type was not successful and another unpacker is used as a fallback). 

The middle error you posted is puzzling, though. It seems to be the reason why your firmware analysis hangs but I cannot tell from this log what exactly may have caused this. It is a pretty generic OS error and it is hard to tell what may have lead to it. It could be related to this: https://bugs.python.org/issue28326. What python version are you using? Does it happen again when you analyze the same file with the same plugin?

chenjianquan7, 2023-08-14:
i am use python 3.8

jstucke, 2024-11-13:
There was a problem with the CWE checker plugin that was fixed in https://github.com/fkie-cad/FACT_core/pull/1156
To be more precise, there was no memory limit for the docker container in which the analysis runs. This could have resulted in out of memory errors when analyzing very large binaries. In hindsight, this may be related to this issue.