fkie-cad/FACT_core #1044: mportError: cannot import name 'uia_username_mapper' from 'flask_security'

hugooda12, 2023-05-18:
### FACT version

7b0c4ebd0c998a620f1e96d81e15486d3f56c7c9

### Environment

Ubuntu 22.04.2 LTS

### Steps to reproduce

run ./start_all_installed_fact_components
after doing Install (python3 install.py)

### Expeced Behavior

access to http://localhost:5000/

### Installation logs

[2023-05-18 08:48:36][install][INFO]: FACT Installer 1.2
[2023-05-18 08:48:36][common][INFO]: Updating system
[2023-05-18 08:48:39][install][INFO]: Installing apt-transport-https autoconf automake build-essential git libtool python3 python3-dev unzip libfuzzy-dev libmagic-dev
[2023-05-18 08:48:44][common][INFO]: Installing python3 pip
[2023-05-18 08:49:48][install][INFO]: Installing nodejs papirus-icon-theme
[2023-05-18 08:51:19][install][WARNING]: Pip package flask-login==0.6.2 is already installed with distutils. This may Cause problems:
Collecting flask-login==0.6.2
  Downloading Flask_Login-0.6.2-py3-none-any.whl (17 kB)
Collecting Flask>=1.0.4 (from flask-login==0.6.2)
  Downloading Flask-2.3.2-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.9/96.9 kB 2.3 MB/s eta 0:00:00
Requirement already satisfied: Werkzeug>=1.0.1 in /usr/local/lib/python3.10/dist-packages (from flask-login==0.6.2) (2.3.4)
Collecting Jinja2>=3.1.2 (from Flask>=1.0.4->flask-login==0.6.2)
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 6.0 MB/s eta 0:00:00
Collecting itsdangerous>=2.1.2 (from Flask>=1.0.4->flask-login==0.6.2)
  Downloading itsdangerous-2.1.2-py3-none-any.whl (15 kB)
Collecting click>=8.1.3 (from Flask>=1.0.4->flask-login==0.6.2)
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 4.6 MB/s eta 0:00:00
Collecting blinker>=1.6.2 (from Flask>=1.0.4->flask-login==0.6.2)
  Downloading blinker-1.6.2-py3-none-any.whl (13 kB)
Requirement already satisfied: MarkupSafe>=2.1.1 in /usr/local/lib/python3.10/dist-packages (from Werkzeug>=1.0.1->flask-login==0.6.2) (2.1.2)
Installing collected packages: Jinja2, itsdangerous, click, blinker, Flask, flask-login
  Attempting uninstall: click
    Found existing installation: click 8.0.3
    Uninstalling click-8.0.3:
      Successfully uninstalled click-8.0.3
  Attempting uninstall: blinker
    Found existing installation: blinker 1.4
ERROR: Cannot uninstall 'blinker'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.

[2023-05-18 08:51:20][install][WARNING]: Pip package flask-paginate==2022.1.8 is already installed with distutils. This may Cause problems:
Collecting flask-paginate==2022.1.8
  Downloading flask_paginate-2022.1.8-py2.py3-none-any.whl (7.2 kB)
Collecting Flask (from flask-paginate==2022.1.8)
  Using cached Flask-2.3.2-py3-none-any.whl (96 kB)
Requirement already satisfied: Werkzeug>=2.3.3 in /usr/local/lib/python3.10/dist-packages (from Flask->flask-paginate==2022.1.8) (2.3.4)
Requirement already satisfied: Jinja2>=3.1.2 in /usr/local/lib/python3.10/dist-packages (from Flask->flask-paginate==2022.1.8) (3.1.2)
Requirement already satisfied: itsdangerous>=2.1.2 in /usr/local/lib/python3.10/dist-packages (from Flask->flask-paginate==2022.1.8) (2.1.2)
Requirement already satisfied: click>=8.1.3 in /usr/local/lib/python3.10/dist-packages (from Flask->flask-paginate==2022.1.8) (8.1.3)
Collecting blinker>=1.6.2 (from Flask->flask-paginate==2022.1.8)
  Using cached blinker-1.6.2-py3-none-any.whl (13 kB)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/local/lib/python3.10/dist-packages (from Jinja2>=3.1.2->Flask->flask-paginate==2022.1.8) (2.1.2)
Installing collected packages: blinker, Flask, flask-paginate
  Attempting uninstall: blinker
    Found existing installation: blinker 1.4
ERROR: Cannot uninstall 'blinker'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.

[2023-05-18 08:51:22][install][WARNING]: Pip package flask-security-too==5.2.0 is already installed with distutils. This may Cause problems:
Collecting flask-security-too==5.2.0
[... 173 lines cut ...]
[2023-05-18 09:01:00][backend][INFO]: Installing file_system_metadata plugin.
[2023-05-18 09:01:02][backend][INFO]: Finished installing file_system_metadata plugin.

[2023-05-18 09:01:02][backend][INFO]: Installing cve_lookup plugin.
[2023-05-18 09:02:31][install][ERROR]: Failed to run python3 setup_repository.py:

### Backend logs

<details>
    <summary>fact_main_backend.log</summary>

```
PASTE HERE
```
</details>

### Frontend logs

<details>
    <summary>fact_main_frontend.log</summary>

```
PASTE HERE
```
</details>

cmajane, 2023-05-19:
I'm facing this same issue.

hugooda12, 2023-05-19:
I did the install with Debian 11 and no issue. must be Ubuntu 22.04 specific. Unfortunetly FACT_Core doesn't work with ARM Binaries but that is fine

jstucke, 2023-05-24:
It's a conflict between the python packages installed through apt and the ones FACT tries to install through pip (Flask-Security-Too depends on blinker). We can't simply uninstall the system package because other packages depend on it. The easiest solution is installing FACT in a virtual environment. To do this, create a virtual env, activate it and install FACT as usual. Of course you also need to activate it each time before starting FACT. For more info see https://docs.python.org/3/library/venv.html.

You may need to install the `python3-venv` package under Ubuntu/Debian to be able to create a venv.