fkie-cad/FACT_core #1052: String analysis for a firmware file

keeplearning-rgb, 2023-06-05:
### The FACT version you are using

FACT 4.1-dev

### Your question

After running a firmware scan on a unknown format of the file. (.MOT in this case) There was a conversion of .MOT file to .bin file representing some UIDvalue.bin which in turn gives a HTML dump of string analysis.
When i separately run the string analysis on the .MOT file it just contains the data in the format of HEX or unreadable characters.

I want to know how the tool has achieved the string analysis in a human readable form and what are the steps involved to get the results like that starting from the conversion of .MOT file to .bin file and getting the string analysis.
I am attaching some POCs alongwith here, which shows running a string analysis on the file directly and the string analysis tool is providing after coverting the .MOT file to .bin file.
[image]
[image]

jstucke, 2023-06-05:
Hi, your file seems to be in the [S-Record](https://de.wikipedia.org/wiki/S-Record) format. The FACT extractor has an "unpacker" which will convert the file to raw binary format (which makes the strings readable). You can also use [objcopy](https://ftp.gnu.org/old-gnu/Manuals/binutils-2.12/html_node/binutils_5.html) from binutils to do this conversion on the command line.

keeplearning-rgb, 2023-06-05:
Thanks for the input
Let me check on this and get back to you

keeplearning-rgb, 2023-06-06:
I tried using objcopy but it provides an error that the format is not recognized 
**objcopy: Unable to recognise the format of the input file `APP.MOT'**

I am using objcopy for the first time, maybe doing something wrong here.
Can you provide any inputs on this?
Thanks

jstucke, 2023-06-06:
You can specify the input file format using the `-I` parameter:
```sh
objcopy -I srec -O binary <input_file> <output_file>
```

closed 2024-11-13 by jstucke