fkie-cad/FACT_core #1056: Qemu exec: The uploaded file itself will not be analyzed

Gstalker, 2023-06-12:
### The FACT version you are using

03d336c2821f5627310727e285a35d1b22f66018

### Your question

The uploaded file itself will not be analyzed.
In my understanding, `Firmware` will also be analyzed by all scheduled analyzers like all other `FileObjects`.

In the picture below, I uploaded a ssh under x64 linux and enabled qemu_exec, but this file was not analyzed
[image]

jstucke, 2023-06-12:
`qemu_exec` is a special kind of plugin: it runs on whole file systems/archives. The reason for that is that you usually need dependencies (libraries, configuration files, etc.) which are missing otherwise. The results are only displayed for the individual files by retrieving the results on the fly from the "parent file". An easy "fix" for your use case could be to simply put the file(s) into a ZIP archive before uploading. The plugin is currently only aimed at Linux ELF files, though. It does not include bare metal emulation with qemu-system.

Gstalker, 2023-06-13:
> `qemu_exec` is a special kind of plugin: it runs on whole file systems/archives. The reason for that is that you usually need dependencies (libraries, configuration files, etc.) which are missing otherwise. The results are only displayed for the individual ELF files by retrieving the results on the fly from the "parent file" (the archive/file system). An easy "fix" for your use case could be to simply put the file(s) into a ZIP archive before uploading. The plugin is currently only aimed at Linux ELF files, though. It does not include bare metal emulation with qemu-system.

This solved my doubts, thanks for reply!

closed 2023-06-13 by Gstalker