fkie-cad/FACT_core #1071: it stop at unpacking?

chenjianquan7, 2023-07-04:
### The FACT version you are using

4.1

### Your question

Unpacking did not progress for 19 hours

[image]

jstucke, 2023-07-05:
If you don't provide more information, we will not be able to help you. Are there any errors/traces in the logs?

chenjianquan7, 2023-07-05:
[image]

[2023-07-03 15:14:45][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:47][db_interface_stats][DEBUG]: Updating backend statistics
[2023-07-03 15:14:47][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:49][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:50][common_redis_binding][DEBUG]: analysis_task: New task received: 360 C221 v. 1.2.3.1
Processed Analysis: []
Scheduled Analysis: ['binwalk', 'crypto_hints', 'crypto_material', 'cve_lookup', 'cwe_checker', 'exploit_mitigations', 'interesting_uris', 'ip_and_uri_finder', 'known_vulnerabilities', 'malware_scanner', 'printable_strings', 'users_and_passwords']
[2023-07-03 15:14:50][unpacking_scheduler][DEBUG]: Started Worker on a62406d1d3b680eed06ae466ba1e3161ce1cb3e6a38394ddaf9a39ba73879af9_81465169 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmpjs3fv86b'>)
[2023-07-03 15:14:50][connectionpool][DEBUG]: Starting new HTTP connection (1): localhost:9900
[2023-07-03 15:14:51][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:52][db_interface_stats][DEBUG]: Updating backend statistics
[2023-07-03 15:14:53][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:54][connectionpool][DEBUG]: http://localhost:9900 "GET /start/tmp4pwmxf_b HTTP/1.1" 200 3
[2023-07-03 15:14:55][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 0 / 0[0m
[2023-07-03 15:14:57][unpacking_scheduler][DEBUG]: Started Worker on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmpzsd5098j'>)
[2023-07-03 15:14:57][connectionpool][DEBUG]: Starting new HTTP connection (1): localhost:9901
[2023-07-03 15:14:57][unpacking_scheduler][DEBUG]: Started Worker on c01758e5fb9e1ec0eeb39b73212ff5e140b114ceff989977f3700c2186887976_16777216 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmptga6an9p'>)
[2023-07-03 15:14:57][connectionpool][DEBUG]: http://localhost:9901 "GET /start/tmp0x6c0whb HTTP/1.1" 200 3
[2023-07-03 15:14:57][unpacking_scheduler][DEBUG]: Started Worker on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmpbp90zqm_'>)
[2023-07-03 15:14:57][connectionpool][DEBUG]: Starting new HTTP connection (1): localhost:9903
[2023-07-03 15:14:57][connectionpool][DEBUG]: Starting new HTTP connection (1): localhost:9902
[2023-07-03 15:14:57][db_interface_stats][DEBUG]: Updating backend statistics
[2023-07-03 15:14:57][connectionpool][DEBUG]: http://localhost:9903 "GET /start/tmpcp__mn14 HTTP/1.1" 200 3
[2023-07-03 15:14:57][unpacking_scheduler][DEBUG]: Started Worker on a1e67ccdc476bd8b3fbd0bdc33e5df78dbdf2ffccd975d6fcc1561242a19e3d9_35651584 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmp3u73bfbw'>)
[2023-07-03 15:14:57][unpacking_scheduler][DEBUG]: [93mQueue Length (Analysis/Unpack): 1 / 3[0m
[2023-07-03 15:14:57][connectionpool][DEBUG]: Starting new HTTP connection (1): localhost:9904
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 3: Begin file_type analysis on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 1: Begin file_type analysis on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 3: Finished file_type analysis on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 1: Finished file_type analysis on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 4: Begin file_type analysis on a62406d1d3b680eed06ae466ba1e3161ce1cb3e6a38394ddaf9a39ba73879af9_81465169
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 2: Begin software_components analysis on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 3: Begin printable_strings analysis on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 3: Finished printable_strings analysis on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 2: Finished software_components analysis on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608
[2023-07-03 15:14:58][connectionpool][DEBUG]: http://localhost:9902 "GET /start/tmp9xno81r3 HTTP/1.1" 200 3
[2023-07-03 15:14:58][unpacking_scheduler][DEBUG]: Started Worker on ac23df589f9f99ac872f43141c904886272b2516fc5414dc631e81b8ac64b374_131072000 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmpjs3fv86b'>)
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 7: Begin crypto_material analysis on 3e7cb7c64bdf79f78a1afd98df5ecbf80eb9fbec71c18d9f20907f44a4781627_353
[2023-07-03 15:14:58][PluginBase][DEBUG]: Worker 7: Begin malware_scanner analysis on 73600b649cb33df933694b5f6d509ffdf4f5c78003ac0a93d813c66963ae0bd0_608
[2023-07-03 15:14:58][unpacking_scheduler][DEBUG]: Started Worker on 80d8fe0ca917920cbcb3b7526567aedae08168eed0cd5df2d7c7b5b6387bada0_20276540 (<TemporaryDirectory '/tmp/fact-docker-mount-base-dir/tmpzsd5098j'>)
[2023-07-03 15:14:58][oms][DEBUG]: Starting scan with ClamAV (1/1)
[... cut ...]
[2023-07-03 15:15:01][oms][DEBUG]: '\n----------- SCAN SUMMARY -----------\nInfected files: 0\nTotal errors: 1\nTime: 0.000 sec (0 m 0 s)\nStart Date: 2023:07:03 15:15:01\nEnd Date:   2023:07:03 15:15:01\n'
[2023-07-03 15:15:01][oms][DEBUG]: indicator: ['0']
[2023-07-03 15:15:01][oms][DEBUG]: result: clean
[2023-07-03 15:15:01][PluginBase][DEBUG]: Worker 7: Begin kernel_config analysis on 6289f6a8d7eac94ae481d22f783f16e29cc542b209e05eb85dfb17445ed59012_6
[2023-07-03 15:15:01][oms][DEBUG]: ClamAV 0.103.6

jstucke, 2023-07-05:
The "oms" / malware scanner plugin has not worked correctly for quite a while now. That's why it was removed and is no longer part of FACT. Errors in this plugin should not cause your unpacking/analysis progress to hang, though. That was very likely caused by something else.

chenjianquan7, 2023-07-05:
how to remove oms, i am use 4.1 now, why it is live this

jstucke, 2023-07-05:
Is it 4.1 or 4.1-dev? Did you upgrade from an older version? Did you upgrade using git or a source code tar.gz from the releases? 

Be that as it may, you could simply delete the "oms" folder from `src/plugins/analysis` or not select it during upload.

chenjianquan7, 2023-07-05:
[image]
it return error on terminal

jstucke, 2023-07-05:
It seems the web frontent of FACT still thinks the malware scanner plugin is there, but in the backend it is already gone. Could you try restarting both the backend and the frontend?

Puzzor, 2023-07-15:
I have a similar issue. I think the ```extractor``` was killed or the ``` python3 start_fact_backend``` was killed excitedly, the log is like below, is there a way to debug it? I didn't find any log text in /tmp/fact_backend.log

[2023-07-15 11:18:29][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:19:21][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:20:14][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:21:06][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:21:58][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:22:50][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:23:42][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:24:34][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:25:26][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:26:18][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21
[2023-07-15 11:27:10][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 21

jstucke, 2023-07-24:
There is currently a known bug in FACT's extractor: if the extraction times out after 10 minutes, it should cancel the extraction of that particular file but instead it retries to connect to the extractor API endpoint which causes the extraction to start again. It could be related to that. We are investigating this and working on a solution.