fkie-cad/FACT_core #1076: 100% memory usage rate

Rvelamen, 2023-07-06:
### The FACT version you are using

3.3+4.1

### Your question

Dear author,

I have been scanning the fact_core.zip (the source code of FACT) using FACT, and I've observed that the memory usage consistently reaches 100%. Upon inspecting the issue with tracemalloc, it seems that the Queue is continuously expanding. I am wondering if this could be a case of memory leakage.

Could you please provide some insights into this matter? Your assistance would be highly appreciated.

Thank

note:
16GB/16core/512GB

Rvelamen, 2023-07-07:
The specific document is user_and_password/bin/alnumspace.chr

dorpvom, 2023-07-07:
Hi Rvelamen,

thanks for the investigation. While we would never call FACT a memory safe application we do invest significant effort into reducing leakage like left-open queues and file descriptors. We will stage this experiment on one of our systems as well to observe it ourselves. Just to be clear: You use FACT 4.1 for analysis and analyze the zipped FACT 3.3? And do you actually analyze the FACT sources or do you analyze a built FACT? I was wondering since `/bin` folders are usually populated only after the build.

Best,
Johannes