fkie-cad/FACT_core #1101: FACT_core install failed

goant, 2023-08-10:
### The FACT version you are using

the newest version,V4.1

### Your question

when run ~/FACT_core/src/install.py, in the step Installing the cve_lookup plugin. it takes more than 2 hours...then failed
[image]

then the step Installing the cve_lookup plugin is still here , it takes more than 2 hours ...

jstucke, 2023-08-10:
Hi, I'm sorry to hear that. We recently changed the installation of the cve_lookup plugin because the NVD discontinues the CVE data feeds. Could you try again with the latest master version until we release a new stable version?

goant, 2023-08-11:
this problem I have fixed

goant, 2023-12-29:
hello，this failed due to the cve database install slow, then I not install the cve database and import the database by hand. the other way is to use FACT_docker. 

 
hello, may I ask how you fixed this question?