fkie-cad/FACT_core #1136: How can I see the logs &  analyze the big files?

oguzhanozgur, 2023-10-06:
### The FACT version you are using

3.1

### Your question

I installed the FACT with Vagrant. I have 2 problems right now.
1. How can I see the logs of the tool ? (for example : how can I understand the analysis is completed?)
2. When I try to upload a big file (like 200MB) and wait a bit long, there is still no analysis.

jstucke, 2023-10-06:
Hi, are you certain you are using version 3.1 and not 4.1?

> How can I see the logs of the tool ?

You can look at the logs (only warnings and errors) in the web GUI (Admin -> Logs in the nav bar). It should also be possible to connect to vagrant using ssh (the logs are created in the /tmp folder by default).

> (for example : how can I understand the analysis is completed?)

You can either use the "system health" page (Info -> System in the nav bar) to show the current status of any running analyses or use the /status REST endpoint (see https://github.com/fkie-cad/FACT_core/wiki/Rest-API#reststatus)

> When I try to upload a big file (like 200MB) and wait a bit long, there is still no analysis.

You will only be able to monitor the analysis progress after the uploaded file was unpacked. The initial unpacking of large files may take some time and may even time out (10 minutes by default). You should find something in the logs in case something goes wrong, though. Please also make sure that you aren't running out of RAM when analyzing large files.

oguzhanozgur, 2023-10-06:
Thanks for your fast reply.

1.Yes, when i click the home button there is a text like "powered by FACT 3.1".

2.When i enter the "system health" page there is a "Pending items for extraction" in backend status. Should it be "0" when the analyze finished?

3.So, should i wait more for bigger files? To avoid timeout, can I increase the default timeout?

jstucke, 2023-10-09:
The version is really old. Did you use the Vagrantfile from https://github.com/botlabsDev/FACTbox? Sadly, it is out of date now. There are pre-built vagrant boxes available with the latest version [here](https://app.vagrantup.com/fact-cad/boxes/FACT-master), though. 
There was a problem with bigger files in the past. I would recommend you try out the new version where it should hopefully be fixed.

> 2.When i enter the "system health" page there is a "Pending items for extraction" in backend status. Should it be "0" when the analyze finished?

In the current version there should be a detailed overview of firmware that is currently undergoing unpacking/analysis on that page (I'm not entirely sure what was visible in version 3.1)

> 3.So, should i wait more for bigger files? To avoid timeout, can I increase the default timeout?

I don't think unpacking should take that long (the default timeout is 10 minutes) even for files that large, so there is probably a different issue. Debugging it doesn't really make a lot of sense, though, since the version you are using is more than 3 years old.

oguzhanozgur, 2023-10-09:
I'm new with this tool. Can you tell how to upgrade the tool step by step?
Thanks in advance.

jstucke, 2023-10-09:
I'm actually not really sure if it is possible to upgrade the existing vagrant box. In general, you can navigate to FACT's installation folder, run `git pull` to fetch the latest changes and rerun the installation (see [INSTALL.md](https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.md)). If you want to try upgrading, you also need to migrate the database (FACT switched from MongoDB to PostgreSQL with 4.x) which is documented [here](https://fkie-cad.github.io/FACT_core/migration.html).

The installation/upgrade process for the new vagrant boxes is described [here](https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.vagrant.md). There is probably no trivial way to migrate data from your old FACT instance into the new vagrant box, though.

oguzhanozgur, 2023-10-09:
When I run the "~/FACT_core/src/install.py", it gives an error like this:
Can you help me? 

#7 4.678 Preparing to unpack .../libpython3.9-minimal_3.9.2-1_amd64.deb ...
#7 4.680 Unpacking libpython3.9-minimal:amd64 (3.9.2-1) ...
#7 4.736 Selecting previously unselected package python3.9-minimal.
#7 4.736 Preparing to unpack .../python3.9-minimal_3.9.2-1_amd64.deb ...
#7 4.740 Unpacking python3.9-minimal (3.9.2-1) ...
#7 4.840 Setting up libpython3.9-minimal:amd64 (3.9.2-1) ...
#7 4.848 Setting up python3.9-minimal (3.9.2-1) ...
#7 4.905 Traceback (most recent call last):
#7 4.905   File "/usr/lib/python3.9/py_compile.py", line 215, in <module>
#7 4.905     sys.exit(main())
#7 4.905   File "/usr/lib/python3.9/py_compile.py", line 207, in main
#7 4.905     compile(filename, doraise=True)
#7 4.905   File "/usr/lib/python3.9/py_compile.py", line 172, in compile
#7 4.905     importlib._bootstrap_external._write_atomic(cfile, bytecode, mode)
#7 4.905   File "<frozen importlib._bootstrap_external>", line 126, in _write_atomic
#7 4.905 PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.9/__pycache__/__future__.cpython-39.pyc.140459029917968'
#7 4.907 dpkg: error processing package python3.9-minimal (--configure):
#7 4.907  installed python3.9-minimal package post-installation script subprocess returned error exit status 1
#7 4.912 Errors were encountered while processing:
#7 4.912  python3.9-minimal
#7 4.917 E: Sub-process /usr/bin/dpkg returned an error code (1)
#7 ERROR: executor failed running [/bin/sh -c apt-get update -y     && apt-get install -y --no-install-recommends python3     && apt-get clean     && rm -rf /var/lib/apt/lists/*]: exit code: 100
------
 > [3/5] RUN apt-get update -y     && apt-get install -y --no-install-recommends python3     && apt-get clean     && rm -rf /var/lib/apt/lists/*:
4.905     compile(filename, doraise=True)
4.905   File "/usr/lib/python3.9/py_compile.py", line 172, in compile
4.905     importlib._bootstrap_external._write_atomic(cfile, bytecode, mode)
4.905   File "<frozen importlib._bootstrap_external>", line 126, in _write_atomic
4.905 PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.9/__pycache__/__future__.cpython-39.pyc.140459029917968'
4.907 dpkg: error processing package python3.9-minimal (--configure):
4.907  installed python3.9-minimal package post-installation script subprocess returned error exit status 1
4.912 Errors were encountered while processing:
4.912  python3.9-minimal
4.917 E: Sub-process /usr/bin/dpkg returned an error code (1)
------
ERROR: failed to solve: executor failed running [/bin/sh -c apt-get update -y     && apt-get install -y --no-install-recommends python3     && apt-get clean     && rm -rf /var/lib/apt/lists/*]: exit code: 100

Traceback (most recent call last):
  File "/home/vestel/FACT_core/src/install.py", line 210, in <module>
    install()
  File "/home/vestel/FACT_core/src/install.py", line 178, in install
    install_fact_components(args, distribution, none_chosen, skip_docker)
  File "/home/vestel/FACT_core/src/install.py", line 198, in install_fact_components
    backend(skip_docker, distribution)
  File "/home/vestel/FACT_core/src/install/backend.py", line 46, in main
    _install_plugins(distribution, skip_docker)
  File "/home/vestel/FACT_core/src/install/backend.py", line 115, in _install_plugins
    plugin_installer.install()
  File "/home/vestel/FACT_core/src/plugins/installer.py", line 56, in install
    self.install_docker_images()
  File "/home/vestel/FACT_core/src/plugins/analysis/software_components/install.py", line 24, in install_docker_images
    self._build_docker_image('fact/format_string_resolver', '../ipc/docker')
  File "/home/vestel/FACT_core/src/plugins/installer.py", line 109, in _build_docker_image
    run_cmd_with_logging(f'docker build -t {tag} {dockerfile_path}')
  File "/home/vestel/FACT_core/src/helperFunctions/install.py", line 217, in run_cmd_with_logging
    raise err
  File "/home/vestel/FACT_core/src/helperFunctions/install.py", line 212, in run_cmd_with_logging
    subprocess.run(cmd_, stdout=PIPE, stderr=STDOUT, encoding='UTF-8', shell=shell, check=True, **kwargs)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['docker', 'build', '-t', 'fact/format_string_resolver', '../ipc/docker']' returned non-zero exit status 1.

jstucke, 2023-10-09:
You are trying to update inside the existing vagrant VM, right? Hmm the first error seems to be a permission problem. Maybe the  user inside the VM does not have the required permissions to install anything. Do you have many existing results in the database that you want to keep? If not installing the new version directly is probably easier and less error-prone.

oguzhanozgur, 2023-10-09:
No, im not trying to install with vagrant. I'm only applying TL;DR steps on https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.md.

jstucke, 2023-10-09:
> No, im not trying to install with vagrant.

But it is still the VM that you originally set up using vagrant? Or is it a local installation?
I'm not sure what could have caused the first error except the user running the installation not having the necessary permissions. Can you manually install `python3.9-minimal` (i.e. `sudo apt install python3.9-minimal`) or does that cause the same error?

oguzhanozgur, 2023-10-11:
When I try to install manually it gives error like this:

/FACT_core$ sudo apt install python3.9-minimal
[sudo] password for vestel: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3.9-minimal
E: Couldn't find any package by glob 'python3.9-minimal'

jstucke, 2023-10-11:
> E: Couldn't find any package by glob 'python3.9-minimal'

I just copied the package name from the log you posted earlier, so it should probably exist (except that log was from a different system):

> https://github.com/fkie-cad/FACT_core/issues/7 4.736 Preparing to unpack .../python3.9-minimal_3.9.2-1_amd64.deb ...

I'm not sure if I understood this correctly: Are you still trying to update your existing installation inside the vagrant VM or are you trying to do a new installation of FACT in your local system?

> https://github.com/fkie-cad/FACT_core/issues/7 4.905 PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.9/pycache/future.cpython-39.pyc.140459029917968'

This sounds like a common error that can occur if you use Docker installed with snap (please make sure you don't have docker installed with snap; docker is set up when you run `pre_install.sh`)

oguzhanozgur, 2023-10-12:
> > E: Couldn't find any package by glob 'python3.9-minimal'
> 
> I just copied the package name from the log you posted earlier, so it should probably exist (except that log was from a different system):
> 
> > https://github.com/fkie-cad/FACT_core/issues/7 4.736 Preparing to unpack .../python3.9-minimal_3.9.2-1_amd64.deb ...
> 
> I'm not sure if I understood this correctly: Are you still trying to update your existing installation inside the vagrant VM or are you trying to do a new installation of FACT in your local system?
>  I'm trying to do a new installation.
> > https://github.com/fkie-cad/FACT_core/issues/7 4.905 PermissionError: [Errno 13] Permission denied: '/usr/lib/python3.9/pycache/future.cpython-39.pyc.140459029917968'
> 
> This sounds like a common error that can occur if you use Docker installed with snap (please make sure you don't have docker installed with snap; docker is set up when you run `pre_install.sh`)
So, what's your suggestion? I stuck with this error.

jstucke, 2023-10-12:
> So, what's your suggestion? I stuck with this error.

Well do you have Docker installed with Snap? You can find out by running `snap list`. If docker is listed there, it is installed with snap. And if so, do you have any docker containers that you don't want to lose/rebuilt in there? 

If no, you can remove docker with `sudo snap remove docker`, reboot and rerun the FACT installation (as documented in the [INSTALL.md](https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.md)) (it will be reinstalled when you run `pre_install.sh`)

If yes, then it would be better to install FACT in a VM (it will not work with docker installed in snap). The simplest way to do this is creating a new VM with Ubuntu 22.04 e.g. with VirtualBox, cloning FACT and then following the [INSTALL.md](https://github.com/fkie-cad/FACT_core/blob/master/INSTALL.md).

oguzhanozgur, 2023-10-12:
Hi again,

I removed the docker and start installation again. I attached the logs.
The problem is when I enter "localhost:5000", there is only "Internal Server Error" string on the window.
[fact_installation.log](https://github.com/fkie-cad/FACT_core/files/12882390/fact_installation.log)

jstucke, 2023-10-12:
It seems you are not using a virtual environment. It is highly recommended to use a venv to avoid exactly that: conflicts with system packages. You can create a venv, activate it and rerun install.py (should go much faster this time, since mostly pip dependencies are installed)

Docs on venv: https://docs.python.org/3/library/venv.html

oguzhanozgur, 2023-10-13:
It works! FACT 4.2-dev version is now installed. 
In this version how can I see the status of analyze? How can I understand it finished?
Also, how much bigger files can I analyze now?

jstucke, 2023-10-13:
That's great to hear! 

> In this version how can I see the status of analyze?

You have two options:
1. Using the web GUI: You can go to the "system health" page (in the navbar: Info -> System). The status of currently running analyses is displayed here under "Currently analyzed firmware"
2. Using the REST API: The rest endpoint `/rest/status` also provides you with this information (see https://github.com/fkie-cad/FACT_core/wiki/Rest-API#reststatus)

> Also, how much bigger files can I analyze now?

That is hard to tell. It depends a lot on how large the files are that you want to analyze, how much RAM you have and how you configure FACT (you can configure FACT in a way that it uses less RAM is needed but the analysis progress will be slower)

jstucke, 2024-02-19:
Since this has been stale for a while, I will close it. If this are still problems or if you have further questions feel free to reopen it.

closed 2024-02-19 by jstucke