fkie-cad/FACT_core #1137: Exception during analysis cve_lookup？

chenjianquan7, 2023-10-08:
### FACT version

4.1

### Environment

[image]

### Steps to reproduce

analysis

### Expeced Behavior

ok

### Installation logs

<details>
    <summary>install.log</summary>

```
PASTE HERE
```
</details>

### Backend logs

<details>
    <summary>fact_main_backend.log</summary>

```
PASTE HERE
```
</details>

### Frontend logs

<details>
    <summary>fact_main_frontend.log</summary>

```
PASTE HERE
```
</details>

chenjianquan7, 2023-10-08:
[image]

the process stop

jstucke, 2023-10-09:
With #1138, analysis exception traces should also be logged to log files and the web GUI