fkie-cad/FACT_core #1145: Issue when uploading large files: invalid HTTP request size (max 4096)

paulrichardo, 2023-10-17:
### The FACT version you are using

a31382a00173fa54df32dc73411833d38540cceb

### Your question

Hello,
I'm starting the server with:
~/Desktop/FACT/FACT_core$ ./start_all_installed_fact_components

All runs fine. I use python requests library with FACT's REST API.
When I query for
http://localhost:5000/rest/status
I get fine response and I see in the window of "start_all_installed_fact_components" that my GET request was processed fine.
Though when I initiate a PUT request to
http://localhost:5000/rest/firmware
with a file of 15MB in size I get 
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))
in python and I see in "start_all_installed_fact_components" window an error:
invalid HTTP request size (max 4096)

After some research I understand that the problem is with uWSGI that it gets large requests. So, I used the 'uwsgi_config.ini' to set the 'buffer-size=65535' to the maximum. But again I have the same error, with the new buffer size. Then I tried to change the nginx.conf settings, but nothing seems to help.

How can I fix this error for large files?

jstucke, 2023-10-17:
Hi and thank you for reporting this. I tried to reproduce your error but without success. Do you have nginx enabled? Could you try it again with this script and see if you get the same error? I submitted a 64 MB file without issues with the script.

```python3
#!/usr/bin/env python3
from __future__ import annotations

from argparse import ArgumentParser
from base64 import b64encode
from pathlib import Path

import requests

def parse_args():
    parser = ArgumentParser(description="FACT REST firmware upload script")
    parser.add_argument("file", type=Path, help="the file to upload")
    parser.add_argument("--device_class", "-c", type=str, help="the device class", default="test_class")
    parser.add_argument("--device_name", "-n", type=str, help="the device name", default="test_device")
    parser.add_argument("--vendor", "-v", type=str, help="the device name", default="test_vendor")
    parser.add_argument("--version", "-V", type=str, help="the device name", default="1")
    return parser.parse_args()

def main():
    args = parse_args()
    data: dict[str, str | list[str]] = {
        "binary": b64encode(args.file.read_bytes()).decode(),
        "device_class": args.device_class,
        "device_name": args.device_name,
        "device_part": "complete",
        "file_name": args.file.name,
        "release_date": "1970-01-01",
        "requested_analysis_systems": [],
        "vendor": args.vendor,
        "version": args.version,
    }
    response = requests.put("http://localhost:5000/rest/firmware", json=data)
    print(f"{response.status_code=}")
    print(f"{response.content=}")

if __name__ == '__main__':
    main()

```

paulrichardo, 2023-10-18:
While you were right about nginx, I reinstalled it and now it works.
But I have another problem.
While the PUT request returns status 200 I get this exception in the terminal:
Traceback (most recent call last):
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/unpacking_scheduler.py", line 186, in _work_thread_wrapper
    self.work_thread(task, container)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/unpacking_scheduler.py", line 208, in work_thread
    db_interface.add_object(task)  # save FO before submitting to analysis scheduler
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/storage/db_interface_backend.py", line 32, in add_object
    self.update_object(fw_object)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/storage/db_interface_backend.py", line 152, in update_object
    self.update_firmware(fw_object)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/storage/db_interface_backend.py", line 158, in update_firmware
    entry.release_date = firmware.release_date
AttributeError: 'NoneType' object has no attribute 'release_date'

I Use the 'release_date' from the script you provided:
"release_date": "1970-01-01"

jstucke, 2023-10-18:
This is a known problem in FACT: You cannot upload an object to FACT that is already in the database as part of another firmware.
There are two kinds of objects in FACT: firmware and file objects. Firmware objects represent firmware containers that were uploaded to FACT and file objects represent the files that were unpacked from those firmware objects. What happened in your trace is the following: you uploaded a new firmware and `db_interface.add_object()` gets called to store it in the DB. But if it is already in the DB, it will only be updated. But when trying to update the object, it cannot be loaded from the DB because it is a "file object" and not a firmware. Sadly, there is no trivial fix for this and there is currently no workaround for this problem other than not uploading parts of a firmware that you previously uploaded.

paulrichardo, 2023-10-18:
Thank you for your time.

closed 2023-10-18 by paulrichardo