fkie-cad/FACT_core #1147: After uploading firmware with REST API error: KeyError: 'unpacker'

paulrichardo, 2023-10-19:
### The FACT version you are using

6cb9354d6ea5160bc56374485db42608ac1b4424

### Your question

Hello,
After uploading ASUS Router fw for a test (zip file from their site) I get an error:
[2023-10-19 09:57:57][unpacking_scheduler][INFO]: Unpacking completed: 51d4c4b657e792ddd918b2d9e927936ede73362b6322d8a6ef6ec76e46669bda_18481152 (extracted files: 1)
[2023-10-19 09:57:57][unpacking_scheduler][ERROR]: Exception occurred during unpacking of 51d4c4b657e792ddd918b2d9e927936ede73362b6322d8a6ef6ec76e46669bda_18481152
Traceback (most recent call last):
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/unpacking_scheduler.py", line 186, in _work_thread_wrapper
    self.work_thread(task, container)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/unpacking_scheduler.py", line 209, in work_thread
    self.post_unpack(task)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/analysis/scheduler.py", line 189, in start_analysis_of_object
    self.task_scheduler.schedule_analysis_tasks(fo, fo.scheduled_analysis, mandatory=True)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/task_scheduler.py", line 22, in schedule_analysis_tasks
    scheduled_analysis = self._add_dependencies_recursively(copy(scheduled_analysis) or [])
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/task_scheduler.py", line 60, in _add_dependencies_recursively
    new_dependencies = self.get_cumulative_remaining_dependencies(scheduled_analyses_set)
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/task_scheduler.py", line 67, in get_cumulative_remaining_dependencies
    return {
  File "/home/ubuntu/Desktop/FACT/FACT_core/src/scheduler/task_scheduler.py", line 68, in <setcomp>
    dependency for plugin in scheduled_analyses for dependency in self.plugins[plugin].DEPENDENCIES
KeyError: 'unpacker'

And the interface doesn't continue the analysis:
[image]

When I upload the same file through the web interface it works fine.

jstucke, 2023-10-19:
did you add "unpacker" to the "requested_analysis_systems"? You don't need to do that. The "unpacker" is not a "real" analysis plugin. Instead, unpacking always happens before analysis

jstucke, 2023-10-19:
But thank you for sharing this problem. The "requested_analysis_systems" should be checked and filtered before the firmware reaches the scheduler, so I would say this is a bug.

jstucke, 2023-10-19:
#1148 should fix this

jstucke, 2023-10-19:
What you can't know is that 'unpacker' is not actually a plugin. It is the output of the unpacking process displayed as if it was a plugin. Otherwise all of the entries should work. But be warned: Some of the plugins can be very slow (e.g. "cwe_checker") and you may not want to run them on every file.

PR #1148 (2023-11-07): rest put firmware: unknown plugin fix
- fixes a bug in route /rest/firmware put when submitting a firmware with unavailable plugins in `'requested_analysis_systems'`
    - the "requested_analysis_systems" were not validated previously, leading to a `KeyError` during scheduling

resolves #1147
closed 2023-10-19 by paulrichardo