fkie-cad/FACT_core #1149: process stop 20hours

chenjianquan7, 2023-10-20:
### The FACT version you are using

4.1

### Your question

[image]
 all process stop, why?
it happend sometimes

chenjianquan7, 2023-10-20:
[image]

chenjianquan7, 2023-10-20:
[image]
[image]
[image]

jstucke, 2023-10-23:
Hi, from what I can tell from the logs it seems like the file you are trying to upload as firmware is already in the database as unpacked file from another firmware (this is currently not possible and may cause errors). Could this be the case?

jstucke, 2023-10-23:
What is more, the file will never leave "currently analyzed firmware" because this file "got lost" during scheduling: All unpacking and analysis tasks should actually be completed but it will not get updated because of the exception. You need to restart FACT to get rid of the entry.

closed 2024-11-19 by jstucke