fkie-cad/FACT_core #1154: why process stop

chenjianquan7, 2023-10-27:
### FACT version

4.1

### Environment

ubuntu

### Steps to reproduce

[image]
why main process stop at 99.8 and other process at over 100?

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 444, in _error_catcher
    yield
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 828, in read_chunked
    self._update_chunk_length()
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 758, in _update_chunk_length
    line = self._fp.fp.readline()
  File "/usr/lib/python3.8/socket.py", line 669, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/requests/models.py", line 816, in generate
    yield from self.raw.stream(chunk_size, decode_content=True)
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 624, in stream
    for line in self.read_chunked(amt, decode_content=decode_content):
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 857, in read_chunked
    self._original_response.close()
  File "/usr/lib/python3.8/contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.8/dist-packages/urllib3/response.py", line 449, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
urllib3.exceptions.ReadTimeoutError: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/chenjianquan/FACT_core-0829/src/helperFunctions/process.py", line 87, in run
    Process.run(self)
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/chenjianquan/FACT_core-0829/src/analysis/PluginBase.py", line 201, in process_next_object
    finished_task = self.analyze_file(task)
  File "/home/chenjianquan/FACT_core-0829/src/analysis/PluginBase.py", line 164, in analyze_file
    fo = self.process_object(file_object)
  File "/home/chenjianquan/FACT_core-0829/src/plugins/analysis/cwe_checker/code/cwe_checker.py", line 143, in process_object
    file_object = self._do_full_analysis(file_object)
  File "/home/chenjianquan/FACT_core-0829/src/plugins/analysis/cwe_checker/code/cwe_checker.py", line 115, in _do_full_analysis
    output = self._run_cwe_checker_in_docker(file_object)
  File "/home/chenjianquan/FACT_core-0829/src/plugins/analysis/cwe_checker/code/cwe_checker.py", line 76, in _run_cwe_checker_in_docker
    result = run_docker_container(
  File "/home/chenjianquan/FACT_core-0829/src/helperFunctions/docker.py", line 43, in run_docker_container
    response = container.wait(timeout=timeout)
  File "/usr/local/lib/python3.8/dist-packages/docker/models/containers.py", line 513, in wait
    return self.client.api.wait(self.id, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/docker/utils/decorators.py", line 19, in wrapped
    return f(self, resource_id, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/container.py", line 1320, in wait
    res = self._post(url, timeout=timeout, params=params)
  File "/usr/local/lib/python3.8/dist-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/docker/api/client.py", line 233, in _post
    return self.post(url, **self._set_request_timeout(kwargs))
  File "/usr/local/lib/python3.8/dist-packages/requests/sessions.py", line 637, in post
    return self.request("POST", url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.8/dist-packages/requests/sessions.py", line 747, in send
    r.content
  File "/usr/local/lib/python3.8/dist-packages/requests/models.py", line 899, in content
    self._content = b"".join(self.iter_content(CONTENT_CHUNK_SIZE)) or b""
  File "/usr/local/lib/python3.8/dist-packages/requests/models.py", line 822, in generate
    raise ConnectionError(e)
requests.exceptions.ConnectionError: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out.

### Expeced Behavior

100%finish

### Installation logs

<details>
    <summary>install.log</summary>

```
PASTE HERE
```
</details>

### Backend logs

<details>
    <summary>fact_main_backend.log</summary>

```
PASTE HERE
```
</details>

### Frontend logs

<details>
    <summary>fact_main_frontend.log</summary>

```
PASTE HERE
```
</details>

jstucke, 2023-10-27:
This seems to be just a timeout of an analysis process. The reason why you don't reach 100% is a different one: the progress of unpacking is also not at 100%. This means that something got lost during unpacking. There was probably some unhandled exception during unpacking which resulted in 2 files being dropped from scheduling. Did you see any other exceptions in the log?  

Also you are not using the latest FACT release version. I suggest that you either switch to 4.1.1 (which contains many bug fixes that 4.1 does not have) or the latest master version.

chenjianquan7, 2023-10-30:
[image]
[image]
[image]
[image]

it stop 30mins

jstucke, 2023-10-30:
All the errors that you posted should be handled in [this line](https://github.com/fkie-cad/FACT_core/blob/e0dc06dfea7b31160c636eb96f1b2729fae3c8d2/src/analysis/PluginBase.py#L210C16-L210C16), so it there are no errors outside of the plugin or the `process_next_object` function, this is not the reason why it will not finish. The `cwe_checker` can be very slow for large binaries, so it is generally advisable to only run it on manually selected files (and not the whole firmware). It could very well be that the analysis is still running after 90 minutes. If you see a blue turning cogwheel below the "current analyses" like this

[image]

then it is still running.

Incidentally, there seems to be a bug in the `cwe_checker` plugin that lets it use unlimited amounts of system memory (which may also be related to your problems). #1156 should fix this.