fkie-cad/FACT_core #1158: The server is not working. (Internal Server Error)

mksonzien, 2023-10-31:
### Your question

[image]

After a fresh installation of Ubuntu, I installed FACT_core, but it does not run.
What can I try?

[error log]
$ sudo ./start_all_installed_fact_components                                      26.5s  화 15:40:48
[2023-10-31 15:40:49][start_all_installed_fact_components][INFO]: starting database
[2023-10-31 15:40:49][start_all_installed_fact_components][INFO]: starting frontend
[2023-10-31 15:40:49][start_all_installed_fact_components][INFO]: starting backend
[2023-10-31 15:40:50][__init__][INFO]: Alembic DB revision:  head: 70ae1212bc03, current: 70ae1212bc03
[2023-10-31 15:40:50][__init__][INFO]: Alembic DB revision:  head: 70ae1212bc03, current: 70ae1212bc03
[2023-10-31 15:40:50][fact_base][INFO]: Successfully started FACT DB-Service
/usr/lib/python3/dist-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
[2023-10-31 15:40:50][__init__][INFO]: Alembic DB revision:  head: 70ae1212bc03, current: 70ae1212bc03
[2023-10-31 15:40:51][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/ip_rules.yara
[2023-10-31 15:40:51][ip_and_uri_finder_analysis][INFO]: ip signature path: /usr/local/lib/python3.10/dist-packages/common_analysis_ip_and_uri_finder/yara_rules/uri_rules.yara
[2023-10-31 15:40:52][compare][INFO]: Comparison plugins available: File_Coverage, File_Header, Software
[2023-10-31 15:40:52][scheduler][INFO]: Analysis scheduler online
[2023-10-31 15:40:52][scheduler][INFO]: Analysis plugins available: binwalk 0.5.5, cpu_architecture 0.4.0, crypto_hints 0.1.1, crypto_material 0.5.2, cve_lookup 0.1.0, cwe_checker 0.5.3, device_tree 1.0.1, elf_analysis 0.3.4, exploit_mitigations 0.1.6, file_hashes 1.2, file_system_metadata 1.0.0, file_type 1.0.0, hardware_analysis 0.2, hashlookup 0.1.4, information_leaks 0.1.4, init_systems 0.4.2, input_vectors 0.1.2, interesting_uris 0.1, ip_and_uri_finder 0.4.2, ipc_analyzer 0.1.1, kernel_config 0.3.1, known_vulnerabilities 0.2.1, printable_strings 0.3.4, qemu_exec 0.5.2, software_components 0.4.2, source_code_analysis 0.7.1, string_evaluator 0.2.1, tlsh 0.2, users_and_passwords 0.5.4
[2023-10-31 15:40:52][extraction_container][INFO]: Started unpack worker 0
[2023-10-31 15:40:53][extraction_container][INFO]: Started unpack worker 1
[2023-10-31 15:40:53][extraction_container][INFO]: Started unpack worker 2
[2023-10-31 15:40:53][extraction_container][INFO]: Started unpack worker 3
[2023-10-31 15:40:53][unpacking_scheduler][INFO]: Queue Length (Analysis/Unpack): 0 / 0
[2023-10-31 15:40:53][unpacking_scheduler][INFO]: Unpacking scheduler online
[2023-10-31 15:40:53][comparison_scheduler][INFO]: Comparison scheduler online
[2023-10-31 15:40:53][back_end_binding][INFO]: Intercom online
[2023-10-31 15:40:53][fact_base][INFO]: Successfully started FACT Backend
[2023-10-31 15:40:55][fact_base][INFO]: System memory usage: 12.4%
[2023-10-31 15:40:58][fact_base][INFO]: System memory usage: 12.3%
[2023-10-31 15:41:01][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /home/zien/01.DEVELOP/14.IoT_PENTEST/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.22 (64bit) on [Tue Oct 31 15:41:01 2023] ***
compiled with version: 11.4.0 on 27 October 2023 06:01:44
os: Linux-6.2.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Oct  6 10:23:26 UTC 2
nodename: zien-System-Product-Name
machine: x86_64
clock source: unix
pcre jit disabled
detected number of CPU cores: 16
current working directory: /home/zien/01.DEVELOP/14.IoT_PENTEST/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
uWSGI running as root, you can use --uid/--gid/--chroot options
*** WARNING: you are running uWSGI as root !!! (use the --uid flag) *** 
your processes number limit is 253997
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
probably another instance of uWSGI is running on the same address (127.0.0.1:5000).
bind(): Address already in use [core/socket.c line 769]
[2023-10-31 15:41:06][fact_base][INFO]: System memory usage: 12.3%

jstucke, 2023-10-31:
Hi,

>  probably another instance of uWSGI is running on the same address (127.0.0.1:5000).
bind(): Address already in use [core/socket.c line 769]

this means there is either another instance of FACT or something else blocking the port. Except for that, there are no errors in the log, so I guess the error could be in the log of the other FACT instance that is running. Without the trace of the error I can't really tell what could cause this.

jstucke, 2024-02-19:
Since this has been stale for a while, I will close it. If this is still an issue, feel free to reopen it.

closed 2024-02-19 by jstucke