fkie-cad/FACT_core #1159: VxWorks analysis shows two versions

paulrichardo, 2023-10-31:
### The FACT version you are using

6cb9354d6ea5160bc56374485db42608ac1b4424

### Your question

Hello,
I'm analyzing this firmware:
Schneider Electric BMX EthernetNet/IP Network Module - BMX NOC 0401 1.2

Can be downloaded from:
https://www.se.com/ca/en/product/BMXNOC0401/network-module-modicon-m340-ethernet-ip-and-modbus-tcp-4-x-rj45/

Unfortunately, no direct link.
Click "See all software & firmware" in the "Latest Firmware" section.
-> "Show Previous Versions"
Filename: NOC0401_v102.zip

When Analyzing this file it shows two versions of VxWorks ['6.4', '5.5']:
[image]

File path:
[image]

How come there are two versions?

dorpvom, 2023-11-02:
Hi paulrichardo,

I found that VxWorks is prone to contain multiple references to conflicting versions. I'll have a look at your sample and see if I can find which one is correct. Rule of thumb though is that it's nearly always the more current version, and only some legacy code of an older version is also included.

dorpvom, 2023-11-02:
Yea. The 5.5 comes from a string " Cmd: it is not possible to use pipe while the shell is configured to be VxWorks 5.5 compatible." So version 6.4 is the correct one.
I'll look to find a way to remove such false positives in the future.

closed 2023-11-05 by paulrichardo