fkie-cad/FACT_core #1161: Database error when trying to write to the database

chenjianquan7, 2023-11-01:
### FACT version

4.2dev

### Environment

Ubuntu20

### Steps to reproduce

[image]
[image]
[image]
[image]

### Expeced Behavior

finish

### Installation logs

<details>
    <summary>install.log</summary>

```
PASTE HERE
```
</details>

### Backend logs

<details>
    <summary>fact_main_backend.log</summary>

```
PASTE HERE
```
</details>

### Frontend logs

<details>
    <summary>fact_main_frontend.log</summary>

```
PASTE HERE
```
</details>

wuzhongjie1997, 2024-11-26:
This has been observed in FACT v4.2 (git checkout 4.2) frequently. 
Relevant logs from back-end: 
```
[2024-11-26 08:23:58][db_interface_base][ERROR]: Database error when trying to write to the database: (psycopg2.errors.UntranslatableCharacter) unsupported Unicode escape sequence
LINE 1: ...Constant','CRC32_table']::VARCHAR[], '{}'::JSONB, '{"matches...
                                                             ^
DETAIL:  \u0000 cannot be converted to text.
CONTEXT:  JSON data, line 1: ..."CRC32_table", "strings": [[88904, "$c0", "\u0000...

[SQL: INSERT INTO analysis (uid, plugin, plugin_version, system_version, analysis_date, summary, tags, result) VALUES (%(uid)s, %(plugin)s, %(plugin_version)s, %(system_version)s, %(analysis_date)s, %(summary)s::VARCHAR[], %(tags)s::JSO
NB, %(result)s::JSONB)]
[parameters: {'uid': '12187a033ccd93ed14b278310a189a18dd104803a2931f8881dcd40e7b0ce517_124640', 'plugin': 'crypto_hints', 'plugin_version': '0.2.1', 'system_version': '4.5.0', 'analysis_date': 1732609430.6754472, 'summary': ['CRC32_poly
_Constant', 'CRC32_table'], 'tags': '{}', 'result': '{"matches": [{"meta": {"open_source": null, "software_name": null, "website": null, "date": "2015-05", "author": "_pusher_", "description": "Look for ... (257 characters truncated) ..
. rule": "CRC32_table", "strings": [[88904, "$c0", "\\u0000\\u0000\\u0000\\u0000\\ufffd0\\u0007w,a\\u000e\\ufffdQ\\t\\ufffd\\u0019\\ufffdm\\u0007"]]}]}'}]
(Background on this error at: https://sqlalche.me/e/20/9h9h)
Traceback (most recent call last):
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1967, in _exec_single_context
    self.dialect.do_execute(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/default.py", line 941, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.UntranslatableCharacter: unsupported Unicode escape sequence
LINE 1: ...Constant','CRC32_table']::VARCHAR[], '{}'::JSONB, '{"matches...
                                                             ^
DETAIL:  \u0000 cannot be converted to text.
CONTEXT:  JSON data, line 1: ..."CRC32_table", "strings": [[88904, "$c0", "\u0000...

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/hl/FACT_core/src/storage/db_interface_base.py", line 54, in get_read_write_session
    session.commit()
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 2028, in commit
    trans.commit(_to_root=True)
  File "<string>", line 2, in commit
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/state_changes.py", line 139, in _go
    ret_value = fn(self, *arg, **kw)
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 1313, in commit
    self._prepare_impl()
  File "<string>", line 2, in _prepare_impl
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/state_changes.py", line 139, in _go
    ret_value = fn(self, *arg, **kw)
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 1288, in _prepare_impl
    self.session.flush()
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 4352, in flush
    self._flush(objects)
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 4487, in _flush
    with util.safe_reraise():
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/session.py", line 4448, in _flush
    flush_context.execute()
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/unitofwork.py", line 466, in execute
    rec.execute(self)
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/unitofwork.py", line 642, in execute
    util.preloaded.orm_persistence.save_obj(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/persistence.py", line 93, in save_obj
    _emit_insert_statements(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/orm/persistence.py", line 1048, in _emit_insert_statements
    result = connection.execute(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1418, in execute
    return meth(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/sql/elements.py", line 515, in _execute_on_connection
    return connection._execute_clauseelement(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1640, in _execute_clauseelement
    ret = self._execute_context(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1846, in _execute_context
    return self._exec_single_context(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1986, in _exec_single_context
    self._handle_dbapi_exception(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 2355, in _handle_dbapi_exception
[... 95 lines cut ...]
    return connection._execute_clauseelement(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1640, in _execute_clauseelement
    ret = self._execute_context(
  File "/home/hl/.venv/lib/python3.10/site-packages/sqlalchemy/engine/base.py", line 1846, in _execute_context
    return self._exec_single_context(
[... cut ...] "rule": "BASE64_table", "strings": [[1575072, "$c0", "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789+/"]]}]}'}]
(Background on this error at: https://sqlalche.me/e/20/9h9h)
[2024-11-26 08:29:29][db_interface_backend][ERROR]: Could not store analysis result of crypto_hints on 71aa77261b1b323292f6d05590912868f4498e54a4fcbae9b935d36b373901f7_2031360: Database error when trying to write to the database
[2024-11-26 08:29:30][fact_base][WARNING]: System memory is running low: 83.4%
```

dorpvom, 2024-11-26:
Hi @wuzhongjie1997,

if you find, that this is still valid on master with a refreshed install we will re-open #1162. So make sure to report back.

Thanks!

jstucke, 2025-11-27:
Hi Benedikt, sorry for the late response (I was on vacation). This bug should be fixed on the master branch, but the Vagrant box wasn't updated for quite some time. We are currently looking into that and a newer version should be available soon.

> Does this impact the available analysis results, or can it be safely ignored?

The analysis process of the firmware as a whole should continue without problems, but the individual results where the error occurred are not saved in the database (so these will be lost).

jstucke, 2025-12-15:
I tested it with the mentioned file and can confirm that this triggers an error. 
The problem is that the analysis result of the "crypto_hints" plugin contains a tuple which in turn contains strings with null bytes (and null bytes are not allowed in postgres). Analysis result data is usually sanitized before storing it in the database, but the sanitization method is currently not compatible with tuples. #1433 should fix this bug.