fkie-cad/FACT_core #1167: Two identical files in different firmwares behavior

paulrichardo, 2023-11-07:
### The FACT version you are using

6cb9354d6ea5160bc56374485db42608ac1b4424

### Your question

Hello,
Just wanted to clearly understand how this works.
I understand that identical files in different firmwares are saved under the same FileObject UID in the database.
I have 2 questions:
1. If a file in "Firmware1" is called "environment.bin" and the file with the same hash in "Firmware2" is called "testing.bin". This means that if first I upload "Firmware1" the file name will be "environment.bin" and when I upload "Firmware2" the file name in both firmwares will be "testing.bin"?
2. If I analyze the file in "Firmware1" with plugins "custom_plugin" and "software_components" and the I upload "Firmware2" and I apply plugins: "custom_plugin" with new functionality, meaning the analysis output data should be different, and the "file_type" plugin.
What will happen?
Will I get an updated object after uploading "Firmware2" that will have the analysis of all the plugins:
software_componens
custom_plugin (updated version)
file_type
Or will the analysis will be overwritten with the selected plugins from the "Firmware2":
custom_plugin (updated)
file_type

jstucke, 2023-11-07:
Hi, 

1.) as far is I can tell you are correct. This isn't that much of a problem, though: There is another data structure that contains all entries called `FileObject.virtual_file_path`. The "virtual file path" has the following structure:

```
{
    "parent UID 1": [
        "file path 1",
        "file path 2",
        ...
    ],
    "parent UID 2": [...],
    ...
}
```

where "parent UID" is the UID of a parent (i.e. a file where this file was extracted from) and "file path" represents the path of the file in the parent container (could be a file system, archive or something else). All the file names should be at the end of the file paths.

2.) In general analyses will not be redone if the analysis result is already in the database. To check this, the current version of the plugin is compared to the version of the plugin in the analysis result. So if you increase the version of your custom plugin (after adding new functionality), it will be rerun (and otherwise it won't). You can force a rerun of all selected plugins during an update if you check the box "Force Analysis Update" regardless of the version, though.

paulrichardo, 2023-11-09:
Regarding the first question.
I checked the "virtual_file_path" table and I see that each file_uid (which is the file in question) has a parent_uid, which is an upper level of that file_uid. Let's take an example when there will be already such entry in the database, like:
`parent_uid = 123_123, file_uid=345_345, file_path = /environment.bin`
And I need to add the next entry:
`parent_uid = 123_123, file_uid=345_345, file_path = /testing.bin`
What will happen?

jstucke, 2023-11-09:
The "virtual_file_path" is always updated and not replaced (also when a firmware is deleted). It supports both a file appearing multiple times in the same container ("parent") file and a file appearing in multiple different firmware images (or rather parent files). If you upload another firmware that contains the file, the "virtual_file_path" should be updated and contain both paths. Your example seems to be of the case that a file appears multiple times in the same parent (which should also work fine). However, there is a special case that is currently not supported: The root file of a firmware (the file of the uploaded firmware image itself) may not appear inside other firmware images (otherwise this could cause unexpected bugs).

closed 2023-11-09 by paulrichardo