fkie-cad/FACT_core #1172: How to find extractor  file in docker or my host

yujack008, 2023-11-20:
### Your question

the unpacker module i can see 
```
patool: Extracting /tmp/extractor/input/f620f62db123b3e26a302b84001c9f02f802ae2a3007b04e3924009cc740b3b1_201609133 ...
patool: running /usr/local/bin/7z e -o/tmp/fact_unpack_bn4gerf6 -- /tmp/extractor/input/f620f62db123b3e26a302b84001c9f02f802ae2a3007b04e3924009cc740b3b1_201609133
patool: ... /tmp/extractor/input/f620f62db123b3e26a302b84001c9f02f802ae2a3007b04e3924009cc740b3b1_201609133 extracted to `/tmp/fact_unpack_bn4gerf6'.
```
but how can find /tmp/fact_unpack_bn4gerf6 . i want to see the extractor  file

jstucke, 2023-11-22:
The path `/tmp/extractor/input/f620f62db123b3e26a302b84001c9f02f802ae2a3007b04e3924009cc740b3b1_201609133` exists only inside the extractor container. What is more, it is mounted from a temporary folder which gets deleted after the extraction is finished. So `/tmp/fact_unpack_bn4gerf6` could be the correct folder but it gets deleted directly after the extraction. 

You can also use the extractor container directly (without FACT). See [the readme](https://github.com/fkie-cad/fact_extractor) for reference.

jstucke, 2023-11-22:
Why do you need to find the folder, though? You can also download all files (the file that gets extracted as well as all extracted files) from FACT's web interface.