fkie-cad/FACT_core #1173: upload operation failed

dditxu, 2023-11-21:
### FACT version

master

### Environment

kali 2023 x64
[image]

### Steps to reproduce

when upload an firmware

### Expeced Behavior

upload firmware success

### Installation logs

<details>
    <summary>install.log</summary>

```
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/web_interface/security/decorator.py", line 13, in decorated_view
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/web_interface/components/io_routes.py", line 43, in get_upload
    analysis_plugins = self.intercom.get_available_analysis_plugins()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/intercom/front_end_binding.py", line 35, in get_available_analysis_plugins
    plugin_dict = self.redis.get('analysis_plugins', delete=False)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/storage/redis_interface.py", line 31, in get
    value = self._redis_pop(key) if delete else self.redis.get(key)
                                                ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/commands/core.py", line 1790, in get
    return self.execute_command("GET", name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/client.py", line 1255, in execute_command
    conn = self.connection or pool.get_connection(command_name, **options)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 1442, in get_connection
    connection.connect()
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 704, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.
[2023-11-21 16:47:03][app][ERROR]: Exception on /upload [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 698, in connect
    sock = self.retry.call_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/retry.py", line 46, in call_with_retry
    return do()
           ^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 699, in <lambda>
    lambda: self._connect(), lambda error: self.disconnect(error)
            ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 987, in _connect
    raise err
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 975, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/web_interface/security/decorator.py", line 13, in decorated_view
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/web_interface/components/io_routes.py", line 43, in get_upload
    analysis_plugins = self.intercom.get_available_analysis_plugins()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/intercom/front_end_binding.py", line 35, in get_available_analysis_plugins
    plugin_dict = self.redis.get('analysis_plugins', delete=False)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sword/Desktop/FACT_core/src/storage/redis_interface.py", line 31, in get
    value = self._redis_pop(key) if delete else self.redis.get(key)
                                                ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/commands/core.py", line 1790, in get
    return self.execute_command("GET", name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/client.py", line 1255, in execute_command
    conn = self.connection or pool.get_connection(command_name, **options)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 1442, in get_connection
    connection.connect()
  File "/usr/local/lib/python3.11/dist-packages/redis/connection.py", line 704, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

```
</details>

### Backend logs

<details>
    <summary>fact_main_backend.log</summary>

```
PASTE HERE
```
</details>

### Frontend logs

<details>
    <summary>fact_main_frontend.log</summary>

```
PASTE HERE
```
</details>

jstucke, 2023-11-22:
Judging from the trace it seems that your Redis service is not running. Is Redis installed correctly? Is the service running (you can check with `systemctl status redis-server`)? If installing Redis is problematic for whatever reason, you can also use a docker container.

dditxu, 2023-11-27:
> Judging from the trace it seems that your Redis service is not running. Is Redis installed correctly? Is the service running (you can check with `systemctl status redis-server`)? If installing Redis is problematic for whatever reason, you can also use a docker container.

yes, these services are all running correctly. Maybe it's something else... I think. But i havent' found the reason yet!

jstucke, 2023-11-27:
Maybe Redis is running but on another port. What happens if you try to connect manually from a Python shell?

```python
from redis import Redis
redis = Redis("localhost", 6379)
redis.set("foo", "bar")
redis.get("foo")
```

You can also check the address/port with `sudo netstat -tulpn | grep redis-server`. It should be 127.0.0.1:6379. If it is different, you can set the address/port in FACT's configuration file (src/config/fact-core-config.toml).

jstucke, 2024-02-19:
Since this has been stale for a while, I will close it. If this is still an issue, feel free to reopen it.

closed 2024-02-19 by jstucke