fkie-cad/FACT_core #1180: fact web not start  ,is status 500

xiaoze521, 2023-12-13:
### Your question

ze@ze-virtual-machine:~/FACT_core$ ./start_all_installed_fact_components
[2023-12-13 14:46:27][start_all_installed_fact_components][INFO]: starting database
[2023-12-13 14:46:27][start_all_installed_fact_components][INFO]: starting frontend
[2023-12-13 14:46:28][__init__][INFO]: Alembic DB revision:  head: 70ae1212bc03, current: 70ae1212bc03
[2023-12-13 14:46:28][__init__][INFO]: Alembic DB revision:  head: 70ae1212bc03, current: 70ae1212bc03
[2023-12-13 14:46:28][fact_base][INFO]: Successfully started FACT DB-Service
[2023-12-13 14:46:29][fact_base][INFO]: Successfully started FACT Frontend
[uWSGI] getting INI configuration from /home/ze/FACT_core/src/config/uwsgi_config.ini
*** Starting uWSGI 2.0.22 (64bit) on [Wed Dec 13 14:46:29 2023] ***
compiled with version: 9.4.0 on 12 December 2023 08:05:15
os: Linux-5.15.0-91-generic #101~20.04.1-Ubuntu SMP Thu Nov 16 14:22:28 UTC 2023
nodename: ze-virtual-machine
machine: x86_64
clock source: unix
pcre jit disabled
detected number of CPU cores: 16
current working directory: /home/ze/FACT_core/src
detected binary path: /usr/local/bin/uwsgi
your processes number limit is 257259
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: enabled
uwsgi socket 0 bound to TCP address 127.0.0.1:5000 fd 3
Python version: 3.8.10 (default, Nov 22 2023, 10:22:35)  [GCC 9.4.0]
Python main interpreter initialized at 0x55d4e2481610
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 500256 bytes (488 KB) for 10 cores
*** Operational MODE: preforking+threaded ***
Traceback (most recent call last):
  File "flask_app_wrapper.py", line 26, in <module>
    from web_interface.frontend_main import WebFrontEnd
  File "/home/ze/FACT_core/src/./web_interface/frontend_main.py", line 8, in <module>
    from web_interface.app import create_app
  File "/home/ze/FACT_core/src/./web_interface/app.py", line 4, in <module>
    from flask_security import uia_username_mapper
  File "/home/ze/.local/lib/python3.8/site-packages/flask_security/__init__.py", line 13, in <module>
    from .core import Security, RoleMixin, UserMixin, AnonymousUser, current_user
  File "/home/ze/.local/lib/python3.8/site-packages/flask_security/core.py", line 18, in <module>
    from flask_babelex import Domain
  File "/home/ze/.local/lib/python3.8/site-packages/flask_babelex/__init__.py", line 20, in <module>
    from flask import __request_ctx_stack
ImportError: cannot import name '__request_ctx_stack' from 'flask' (/home/ze/.local/lib/python3.8/site-packages/flask/__init__.py)
unable to load app 0 (mountpoint='') (callable not found or import error)
*** no app loaded. going in full dynamic mode ***
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI master process (pid: 75187)
spawned uWSGI worker 1 (pid: 75244, cores: 2)
spawned uWSGI worker 2 (pid: 75245, cores: 2)
spawned uWSGI worker 3 (pid: 75247, cores: 2)
spawned uWSGI worker 4 (pid: 75249, cores: 2)
spawned uWSGI worker 5 (pid: 75251, cores: 2)

jstucke, 2023-12-13:
It seems you have a very old version of "flask_babelex" installed (the offending line was apparently changed in April 2016) and we did not pin the version. Could you try updating the version by running `python3 -m pip install -U flask_babelex`?

karazonifis, 2023-12-25:
> It seems you have a very old version of "flask_babelex" installed (the offending line was apparently [changed in April [...]

I have the same problem on 2 machines (although I followed the installation instructions strictly), and `python3 -m pip install -U flask_babelex didn't help either.

fuomag9, 2024-01-06:
Same issue here, flask was also missing

jstucke, 2024-01-08:
Hi, 

> Defaulting to user installation because normal site-packages is not writeable

This actually indicates that you are using your system's python without a venv, which is not recommended, since it could cause conflicts with the python packages installed in your system (which could very well be the issue here). Please make sure to set up FACT in its own venv.

If you need some help handling virtual environments, I suggest reading https://docs.python.org/3/library/venv.html

jstucke, 2024-01-08:
If you have run the installation without a venv, it should suffice to 
- set up a venv and activate it
- install the packages from the `pre_install.sh` script (`pip install -r src/install/requirements_pre_install.txt`) 
- run `install.py` again

Of course you also need to activate the venv each time to run FACT

neumannlyu, 2024-06-05:
same problem。
resolve by set python venv and use root user

closed 2024-11-11 by rhelmke