fkie-cad/FACT_core #1195: Schema mismatch right after install

maringuu, 2024-02-16:
### FACT version

4.1.1

### Environment

Ubuntu 20.04.06 LTS

### Steps to reproduce

1) Install FACT
2) Run the backend
3) FACt will complain about a schema mismatch, despite the fact that it is freshly installed

### Expeced Behavior

It works.

### Backend logs

```
[2024-02-16 15:43:36][migration][INFO]: Context impl PostgresqlImpl.
[2024-02-16 15:43:36][migration][INFO]: Will assume transactional DDL.
[2024-02-16 15:43:36][fact_base][ERROR]: The database schema in "storage/schema.py" does not match the schema of the configured database. Please run `alembic upgrade head` from `src` to update the schema.
Traceback (most recent call last):
  File "./start_fact_backend", line 125, in <module>
    FactBackend().main()
  File "./start_fact_backend", line 48, in __init__
    super().__init__()
  File "/home/maringuu/opt/FACT_core-v4.1.1/src/fact_base.py", line 40, in __init__
    self.do_self_test()
  File "/home/maringuu/opt/FACT_core-v4.1.1/src/fact_base.py", line 88, in do_self_test
    raise DbInterfaceError('Schema mismatch')
storage.db_interface_base.DbInterfaceError: Schema mismatch

```

jstucke, 2024-04-26:
@SpaceMoehre are you using FACT 4.1.1 or the master branch?

lrcarp17, 2024-04-30:
I had this issue, it seems like running `DROP TABLE alembic_version;` and re-installing fixed it for me

lrcarp17, 2024-04-30:
However now it seems like I'm getting a 500 error when accessing firmware because the schema has changed. So idk if it's a true fix.