fkie-cad/FACT_core #1198: Expose the uwsgi app

maringuu, 2024-02-19:
Currently the FACT frontend ist started via `./start_fact_frontend`.
This script starts some docker containers, `uwsgi`, and most importantly has some initialisation logic.
This is not the right approach to expose a wsgi app. In short, it should be possible to run `flask --app src/flask_app_wrapper.py run`.
Currently this errors out with the following:
<details>
<summary>Error</summary>

```
Traceback (most recent call last):
  File "/root/venvs/fact-v4.1.1/bin/flask", line 8, in <module>
    sys.exit(main())
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/flask/cli.py", line 1047, in main
    cli.main()
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/decorators.py", line 92, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/flask/cli.py", line 911, in run_command
    raise e from None
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/flask/cli.py", line 897, in run_command
    app = info.load_app()
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/flask/cli.py", line 308, in load_app
    app = locate_app(import_name, name)
  File "/root/venvs/fact-v4.1.1/lib/python3.10/site-packages/flask/cli.py", line 218, in locate_app
    __import__(module_name)
  File "/root/opt/fact-v4.1.1/src/flask_app_wrapper.py", line 41, in <module>
    web_interface = create_web_interface()
  File "/root/opt/fact-v4.1.1/src/flask_app_wrapper.py", line 37, in create_web_interface
    setup_logging(args, 'frontend')
  File "/root/opt/fact-v4.1.1/src/helperFunctions/program_setup.py", line 81, in setup_logging
    logfile, file_loglevel, console_loglevel = _get_logging_config(args, component)
  File "/root/opt/fact-v4.1.1/src/helperFunctions/program_setup.py", line 63, in _get_logging_config
    console_loglevel = logging.getLevelName(args.log_level)
AttributeError: 'NoneType' object has no attribute 'log_level'
```
</details>
Not being able to run the flask debug server has significantly decreased my productivity when working on the frontend.
Besides the argument that the current implementation is unusual and too complicated, the debug server is the main motivation
for this issue.
Another motivation is to not depend on uwsgi but rather let the user decide which wsgi server they want to use.

I think two things have to be done:
- Rethink the logic that prevents it from working currently
- Start the docker containers in another way (read: docker-compose)

jstucke, 2024-02-19:
In general it is possible to start the app directly with your WSGI framework of choice but the app expects an argument containing the path to the parsed CLI args. If you provide it, it works e.g.

```sh
gunicorn flask_app_wrapper:app /tmp/pickle
```

If we change the way how the CLI arguments are handled in the app, we could get it to work without args (e.g. pass the config through Redis or just load a default).

jstucke, 2024-02-19:
this minimal patch seems to be enough for just being able to start the app with default config:

```diff
diff --git a/src/flask_app_wrapper.py b/src/flask_app_wrapper.py
index 434c801e..a710e7a7 100644
--- a/src/flask_app_wrapper.py
+++ b/src/flask_app_wrapper.py
@@ -32,7 +32,9 @@ def create_web_interface():
     if args_path.is_file():
         args = pickle.loads(args_path.read_bytes())
         config_file = getattr(args, 'config_file', None)
-        config.load(config_file)
+    else:
+        config_file = Path(__file__).parent / 'config' / 'fact-core-config.toml'
+    config.load(config_file)
 
     setup_logging(args, 'frontend')
     return WebFrontEnd()
```