fkie-cad/FACT_core #1207: Replace redis by Redict

maringuu, 2024-03-25:
See [here](https://redict.io/posts/2024-03-22-redict-is-an-independent-fork/) for a nice blog article describing what happened to redis.

tldr: Redis relicensed to a non-free (as in freedom) license.

There might emerge more forks of redis, lets see how they evolve and chose the most promising one.

jstucke, 2024-04-02:
Since the deb packages will probably still use the BSD licensed version for the next couple of months, we don't have to take immediate action . I would assume the packages versions for Ubuntu 24.04 are probably also fixed already with a "usable" Redis version. But should keep this in mind and make the switch if necessary

jstucke, 2025-10-24:
Since Redis did a 180 and returned to an OSS license, I will close this for now.