fkie-cad/FACT_core #1239: Add ability to flip compare/diff highlighting

soxrok2212, 2024-07-25:
### Feature description

When I compare `old_firmware` and `new_firmware`, often times the diffing code shows `old_firmware` as having the new changes as indicated by the highlighting. 

It would be handy to be able to reverse this so the diff highlights the changes in the `new_firmware` instead.

jstucke, 2024-07-26:
You mean the (text) diff between two files and not the firmware comparison page, right? One (arguably not very practical) way to achieve that is to switch IDs in the URL. It should be easy to add a button that does that. I will take a look.

soxrok2212, 2024-07-26:
> You mean the (text) diff between two files and not the firmware comparison page, right?
Yes! I actually did try flipping the IDs in the URL and each time it went back to the original.

jstucke, 2024-08-01:
> > You mean the (text) diff between two files and not the firmware comparison page, right?
> 
> Yes! I actually did try flipping the IDs in the URL and each time it went back to the original.

To be more specific:
- It does not work on the comparison page (i.g. `/compare/<uid1>;<uid2>`) 
- but it should work on the text file diff page (i.e. `/comparison/text_files/<uid1>/<uid2>`)

soxrok2212, 2024-08-01:
> To be more specific:
>
> * It does not work on the comparison page (i.g. `/compare/<uid1>;<uid2>`)
> * but it should work on the text file diff page (i.e. `/comparison/text_files/<uid1>/<uid2>`)

Ah, this does work :)