fkie-cad/FACT_core #1242: explaining the analysis😭😭😭

0cookie0, 2024-08-01:
### Your question

I am a novice, who can tell me the meaning of these analysis result entries? Or could you please provide me with a document explaining the analysis results? Thank you very  😭😭😭😭😭😭😭😭😭
[image]

0cookie0, 2024-08-01:
[image]
😭😭😭
[image]

jstucke, 2024-08-01:
> who can tell me the meaning of these analysis result entries?

What you are seeing are not analysis results, this is only a summary for the included files. 

[image]

The analysis result for this file is displayed in the red box (here it is empty). The summary of the analysis results for the included files (the files unpacked from this files) is displayed in the blue box. To view the analysis results, you can click on the link (marked with a pink box). There can be multiple files with the same summary result and to view them all you can click on the button (marked with a green box). To browse the files that were unpacked from this file, you can click on the little triangle left of the file in the file tree (marked with a yellow box). The results are always specific for a single file.

> 😭😭😭
> Uploading 微信截图_2024080111092422.png…

What you marked in red is the list of analysis plugins that were executed for this file. You can click on each entry to view the results of a specific analysis plugin (e.g. what you posted in the first screenshot is the result of the "crypto materials" plugin which tries to find cryptographic material like private keys or certificates).

jstucke, 2024-08-01:
And one more thing:
[image]
If you hover over the list of plugins with your mouse, a small description is displayed as a tooltip

closed 2024-11-11 by rhelmke