fkie-cad/FACT_core #1250: How to import/export firmware database between FACT instances?

ankien, 2024-08-26:
### The FACT version you are using

4.1

### Your question

Hello,

I have an instance of FACT running on a server VM that due to unrelated issues, is not viable to run on anymore. Is there a way to transfer all firmware files and their analysis from that instance to another machine running FACT (preferably a method using the command line)? Or would I be better off uploading all firmware files again to the new instance?

Thanks

jstucke, 2024-08-27:
Though I have not tried it, it should be possible. There are three directories that you need to save:
- The config folder of FACT (only if you changed the configuration, especially passwords)
- the folder `/media/data/fact_fw_data/` which contains all files
- the postgres data folder `/var/lib/postgresql/14/main/`
The best way to do this would probably be to copy the FACT folder, reinstall FACT (first pre_install.sh, then install.py) and then copy over all the data. 

It is also possible to backup and restore the postgres DB with CLI tools: https://www.postgresql.org/docs/current/backup-dump.html

dorpvom, 2024-08-27:
Hi,
just for reference: The import / export [topic](https://github.com/fkie-cad/FACT_core/issues/446) is not new. We haven't yet had a great idea how to solve this. So it remains ..

ankien, 2024-08-27:
> Though I have not tried it, it should be possible. There are three directories that you need to save:
> 
>     * The config folder of FACT (only if you changed the configuration, especially passwords)
> 
>     * the folder `/media/data/fact_fw_data/` (which contains all files)
> 
>     * the postgres data folder `/var/lib/postgresql/14/main/` (which contains all analysis results and meta data)
>       The best way to do this would probably be to copy the FACT folder, reinstall FACT (first pre_install.sh, then install.py) and then copy over all the data.
> 
> 
> It is also possible to backup and restore the postgres DB with CLI tools: https://www.postgresql.org/docs/current/backup-dump.html

So I tried to do this, but when I restart FACT with the data, I get a bunch of SQL errors that look like to this:

`psycopg2.OperationalError: connection to server at "localhost" (127.0.0.1), port 5432 failed: FATAL: could not open file "base/16385/2601": Permission denied`

Looked up similar errors online and I presume this might be fragments from from unfinished analysis, or is it something else?

jstucke, 2024-08-28:
> could not open file "base/16385/2601": Permission denied

That sounds to me like you need to do a `sudo chown -R postgres:postgres /var/lib/postgresql/` because posgres runs under its own user and is not the owner of the newly copied files.

jstucke, 2024-09-05:
So... did it work?

PR #1278 (2024-10-31): FW import/export
- feat: added CLI script for importing/exporting FW (including unpacked files and analysis results)

resolves #1250, resolves #446
closed 2024-10-31 by jstucke