fkie-cad/FACT_core #1267: Option to select which unpacker to use

soxrok2212, 2024-09-17:
### Feature description

It would be useful to be able to select a component and choose which unpacker to use. For example, I have a kernel with an initramfs embedded inside it. I think patool would have no problem extracting it, but 7z is chosen instead by default, which fails.

jstucke, 2024-09-25:
That's an interesting idea. Currently the unpacker is chosen solely based on the detected MIME type, so we would first need to add this as a feature to the extractor. By the way, what MIME type is detected for your file? Do you think it was not detected correctly or do you think patool would generally be better for this type?

soxrok2212, 2024-09-25:
It was definitely detected correctly. Maybe the LinuxImage unpacker missing some logic to search deeper. 

The file is an EFI image (bzImage x86 kernel) and has an embedded initramfs. LinuxUnpacker pulls out 2 EFI files of different sizes, but no initramfs.